arendst/Tasmota #4041: Sonoff B1 Color Temperature

ahmaddxb, 2018-10-12:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

I flashed the Sonoff B1 and have been messing around with it with tasmota. I noticed that the Cold setting doesnt match the Sonoff B1 with its original firmware they look different. Why is that?

ahmaddxb, 2018-10-12:
I flashed the Sonoff B1 and have been messing around with it with tasmota. I noticed that the Cold setting doesnt match the Sonoff B1 with its original firmware they look different. Why is that?

ascillato, 2018-10-12:
Hi, you can use the commands in the console to set the corrections for your case. Please see commands in the wiki. Thanks

closed 2018-10-12 by ascillato2

ahmaddxb, 2018-10-12:
Could you be a bit more clear is can only see command for LedTable on and off, once i turn it on what command do i use to correction?

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

ascillato, 2018-10-12:
For color temperature correction, please try command CT. 

Ledtable is for color correction

ahmaddxb, 2018-10-12:
CT lowest is 153 and that doesnt match sonoff b1 with stock firmware