arendst/Tasmota #4043: Timer setup, how to setup repetition?

Vendo232, 2018-10-12:
Hello

I`m not an expert but I have few Basic Sonoffs around the house hooked on Home Assistant.
I wanted to use simple timer on few of those instead of creating complicated automation.

I have set the right time zone in Console, I have selected ARM when I create ON and OFF setup in Tasmota.

The timer worked ONCE, the next day it did not execute. Would you know where I`m doing mistake?

ascillato, 2018-10-12:
Have you set all the days on? (the list of days of the week is at the bottom of the timers configuration menu)

Vendo232, 2018-10-12:
Yes Sir, I did that. After ONE cycle is done the selection on ARM dissapeers. Not sure what ARM means. but Im having trouble to setup the timer to go ON and OFF the same time. 
[image]

ascillato, 2018-10-12:
Hi, you also need to tick the repeat checkbox in order to be repeated.

ARM means ARMED, that it is that the timer will trigger

closed 2018-10-13 by ascillato2