arendst/Tasmota #4044: Wifi cannot connect Sonoff Basic tasmota ver 6.2 master

balazsfarkas, 2018-10-12:
Hi,

After flash I cannot connect to AP anymore. After the 6.2 master I tried Delevelopment 6.2.1.15-2_3_0 , still the same. I don't have this problem with Sonoff RF.

00:00:00 Project sonoff Sonoff (Topic LED, Fallback DVES_157201, Group Topic sonoffs) Version 6.2.1.15-2_3_0
00:00:00 WIF: Connecting to AP2 Sonoff in mode 11N as LED...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 1 Balazs in mode 11N as LED...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WifiManager active for 3 minutes
00:00:14 HTP: Web server active on LED with IP address 192.168.4.1

Any idea?

Thanks,

ascillato, 2018-10-12:
Hi,

Please complete the troubleshooting template so as to give us the information requires in order to properly help you. Thanks

balazsfarkas, 2018-10-12:
> Hi,
> 
> Please complete the troubleshooting template so as to give us the information requires in order to properly help you. Thanks

Hi, 
Is anything else needed? 
Thanks,

ascillato, 2018-10-12:
Hi, Thanks for complete.

You are on the minimal version. That version is an interim for update (is the first step of the 2 steps OTA flash)

Please, upgrade to the full version.

balazsfarkas, 2018-10-12:
> Hi, Thanks for complete.
> 
> You are on the minimal version. That version is an interim for update (is the first step of the 2 steps OTA flash)
> 
> Please, upgrade to the full version.

Done, same result. 

00:00:15 CMD: status 0
00:00:15 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1
,"ButtonRetain":0,"PowerRetain":0}}
00:00:15 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":
"0T00:00:15","StartupUTC":"","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"FA000"}}
00:00:15 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-12T21:18:42","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}

00:00:15 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Balazs","Sonoff"],"TelePeriod":300,"SetOption":["00008009","55
818000","00000000"]}}
00:00:15 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":470,"Free":532,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","24000
000","23B617CE","00000000"]}}
00:00:15 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4609","IPAddress":"0.0.0.0","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac"
:"B4:E6:2D:15:72:01","Webserver":2,"WifiConfig":5}}
00:00:15 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.240","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_157201","MqttUser":"DVES_USER","MqttType":1
,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:00:15 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:15 1970","Local":"Thu Jan 01 00:00:15 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:0
0 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
00:00:15 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:15"}}
00:00:15 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:15","Uptime":"0T00:00:15","Vcc":3.058,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Balazs","RSSI":100,"APMac":"FF:FF
:FF:FF:FF:FF"}}}

FlorinPBV, 2018-10-12:
Hello, i am just trying to connect a brand new sonoff flashed with the latest sonoff.bin version and i have the same issue. Actually i have the same issue on several of my devices especially after ver. 5.x. My ssid is Bunch Wi-fi and the password is a simple word. The weird part is that after 10 restarts it connects

balazsfarkas, 2018-10-12:
This device (and the other 25 ) is using this WIFI .  This device had tasmota 5.14 since moths, worked well, then I tried to upgrade to 6.2 via OTA. After the update this device did not connect anymore, so I re-flashed with cable, but still the same. 
Now after several flashing and restart connected but, loosing packets, the connection is not stable

Request timeout for icmp_seq 8782
Request timeout for icmp_seq 8783
Request timeout for icmp_seq 8784
Request timeout for icmp_seq 8785
Request timeout for icmp_seq 8786
Request timeout for icmp_seq 8787
Request timeout for icmp_seq 8788
Request timeout for icmp_seq 8789
Request timeout for icmp_seq 8790
Request timeout for icmp_seq 8791
64 bytes from 192.168.1.176: icmp_seq=8782 ttl=128 time=10896.595 ms
64 bytes from 192.168.1.176: icmp_seq=8783 ttl=128 time=9896.714 ms
64 bytes from 192.168.1.176: icmp_seq=8784 ttl=128 time=8893.199 ms
64 bytes from 192.168.1.176: icmp_seq=8785 ttl=128 time=7889.932 ms
64 bytes from 192.168.1.176: icmp_seq=8786 ttl=128 time=6891.301 ms
64 bytes from 192.168.1.176: icmp_seq=8787 ttl=128 time=5894.412 ms
64 bytes from 192.168.1.176: icmp_seq=8788 ttl=128 time=4896.905 ms
64 bytes from 192.168.1.176: icmp_seq=8789 ttl=128 time=3899.047 ms
64 bytes from 192.168.1.176: icmp_seq=8790 ttl=128 time=2901.147 ms
64 bytes from 192.168.1.176: icmp_seq=8791 ttl=128 time=1903.624 ms
64 bytes from 192.168.1.176: icmp_seq=8792 ttl=128 time=901.823 ms
Request timeout for icmp_seq 8803
Request timeout for icmp_seq 8804
Request timeout for icmp_seq 8805
Request timeout for icmp_seq 8806
64 bytes from 192.168.1.176: icmp_seq=8804 ttl=128 time=3847.404 ms
64 bytes from 192.168.1.176: icmp_seq=8805 ttl=128 time=2846.575 ms
64 bytes from 192.168.1.176: icmp_seq=8806 ttl=128 time=1847.364 ms
64 bytes from 192.168.1.176: icmp_seq=8807 ttl=128 time=849.154 ms
Request timeout for icmp_seq 8811
Request timeout for icmp_seq 8812
Request timeout for icmp_seq 8813
Request timeout for icmp_seq 8814
Request timeout for icmp_seq 8815
Request timeout for icmp_seq 8816
Request timeout for icmp_seq 8817
Request timeout for icmp_seq 8818
Request timeout for icmp_seq 8819
Request timeout for icmp_seq 8820
Request timeout for icmp_seq 8821
Request timeout for icmp_seq 8822
Request timeout for icmp_seq 8823

ascillato, 2018-10-12:
Ok,

Lets try another thing.

Try to upload the precompiled bins from the development branch: http://thehackbox.org/tasmota/

Frogmore42, 2018-10-12:
Also completely erase flash before the new upload to clear out old configuration that can make things not work.

balazsfarkas, 2018-10-12:
Erasing then flash again with same result: the device is unreachable, no LED blinking.

esptool.py v2.5.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:15:72:01
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.1s
Hard resetting via RTS pin...

esptool.py v2.5.1
Serial port /dev/tty.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: b4:e6:2d:15:72:01
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0220
Compressed 496640 bytes to 340483...
Wrote 496640 bytes (340483 compressed) at 0x00000000 in 30.1 seconds (effective 132.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

ascillato, 2018-10-12:
which file did you use? sonoff.bin ?

balazsfarkas, 2018-10-12:
> which file did you use? sonoff.bin ?

Yes, that one.

ascillato, 2018-10-12:
Ok,

After you reset your device, there is no AP, no other wifi network as sonoff-xxxx ?

balazsfarkas, 2018-10-12:
> Ok,
> 
> After you reset your device, there is no AP, no other wifi network as sonoff-xxxx ?

I think the esptool did something wrong. I've flashed with that FW what I compiled in Atom, then the device was unreachable. Then uploaded the same FW with Atom then the device come back to life, LED is blinking and I have serial connection. But still no wifi.... So now I'm trying to find what is wrong with esptool...

ascillato, 2018-10-12:
Ok, so may be it is better to continue this chat in the Tasmota Chat at https://discord.gg/Ks2Kzd4

balazsfarkas, 2018-10-12:
> Ok, so may be it is better to continue this chat in the Tasmota Chat at https://discord.gg/Ks2Kzd4

Thanks, I'm there!

balazsfarkas, 2018-10-12:
with 5.12 working fine!

ChrSch84, 2018-10-13:
Looks like the same Problem: https://github.com/arendst/Sonoff-Tasmota/issues/3934 
I must stay on 6.2.1 or 6.2.1.8 all above i got no connection.

andrethomas, 2018-10-13:
I'm not sure if it was covered during the chat session on Discord, but some sonoff basic's seem to have borderline power supply issues so if not already recommended you may want to try the lightened version of the sonoff firmware specifically for the sonoff basic R2 wherein we stripped all the unnecessary drivers from the codebase and compiled it to http://thehackbox.org/tasmota/sonoff-basic_R2.bin

Since a boot with any other firmware may have inadvertently caused garbage to be written to the SPI flash it is recommended to perform a full erase of the SPI flash and then flash the above firmware binary.

Just a shot in the dark ;)

Frogmore42, 2018-10-14:
The other thing to try is see what the serial interface outputs.  The bootloader ALWAYS outputs something.  If you see nothing, you either don't have it hooked up properly or it is not getting power.  Since the Basic has a suspect power supply try powering it with a capable 3V3 supply that is regulated.  A few people have seen it work this way, but not via the built-in power supply.

The serial output will indicate what the software thinks is happening.

ascillato2, 2018-10-14:
Hi, the sdk 3.0.0 from stage core has a different DTIM configuration than the previous core. So, we need to separate the issue that balazsfarkas is having (using core 2.3.0) with the sleep function using core in stage version. This may be not related issues due different cores being used.

Will investigate the issue with core stage. 

DTIM is using 3 by default. Tasmota does not change that.

ascillato, 2018-10-16:
@balazsfarkas 

Please, try this precompiled bin for your sonoff basic.

[firmware.zip](https://github.com/arendst/Sonoff-Tasmota/files/2481138/firmware.zip)

This one is the same Tasmota v6.2.1.15 but with an extra line in support.ino file at L1335:
`WiFi.persistent(false);`

This may solve your wifi issue. Idea from ReloxX13.

balazsfarkas, 2018-10-16:
Will check on this evening , thanks!
Currently the device is working well with 5.12 , so I’m ready to destroy it 😁🤘

ascillato, 2018-10-16:
As now this modification is in the Tasmota codebase, you can upload the precompiled bins from the development branch: http://thehackbox.org/tasmota/

GerritBuwalda, 2018-10-16:
Just tried 6.2.16 on a Sonoff Basic that is experiencing this issue and still no WiFi connection.
Reverting back to 6.2.13 solves this issue.

Last weeks i tried versions 6.2.1, 6.2.11 and 6.2.13 on several Sonoff Basic, Sonoff S20, Coolcam plug and Magichome devices without any problems.

This issue started with version 6.2.14 on Sonoff Basic devices.
Sonoff S20, Coolcam plug and Magichome devices are running fine with 6.2.14.

Hope this info will help solving the problem.

GerritBuwalda, 2018-10-16:
I always compile the bin myself but in this case i used a precompiled bin from the development branch as mentioned above. The precompiled Sonoff-classic.bin (6.2.1.16) run's fine on my Sonoff Basics.

ascillato, 2018-10-16:
@balazsfarkas

Hi,

Have you managed to test it?

ascillato2, 2018-10-17:
@balazsfarkas

Hi,

Have you managed to test it?

timstanley1985, 2018-10-17:
Hi,

I have been having serious issues with WiFi disconnects for about a fortnight, been driving me crazy! I even considered throwing all my sonoffs in the bin having spent all Saturday manually reflashing each one for them just to fall again. Pleased to say I recompiled about 9 hours ago and not a single disconnect since.  Will keep an eye over the next few days and report back

ascillato, 2018-10-17:
Hey, Thanks. Those are really good news. Keep us posted please. Thanks

balazsfarkas, 2018-10-17:
> @balazsfarkas
> 
> Hi,
> 
> Have you managed to test it?

@ascillato Hi, Unfortunately I had no time, sorry - maybe tomorrow evening! Thanks for chasing me!! 👍

timstanley1985, 2018-10-18:
24 hours in and of the 7 sonoff basics I am running 6 have been 100% stable and one had a few disconnects over night but has been stable all day.  Will report back again tomorrow but so far so good!

GerritBuwalda, 2018-10-18:
My 5 production sonoff basics run fine on 6.2.1.16, just upgraded two test sonoff basics to a self compiled 6.2.1.17 bin without any problems.

FlorinPBV, 2018-10-18:
Hello devs, thanks for your quick response and t-shoot. Unfortunately for me the test firmware did not work. I flashed a brand new sonoff with the latest firmware and i have the same errors. In a couple of days i will receive some Cisco APs and i will create a new SSID in my home in order to eliminate all factors.

balazsfarkas, 2018-10-18:
> @balazsfarkas
> 
> Hi,
> 
> Have you managed to test it?

@ascillato Hi, I've flashed that FW but no wifi, even then webconfig is not visible.

balazsfarkas, 2018-10-18:
With the last 6.2.1.17 dev version connect immediately to the WIFI  , but on the web interface I often lost the connection.  When it works really responsive, but sometimes I need to refresh the browser windows because it said page not found.

20:40:53 CMD: status 0
20:40:53 MQT: stat/LED/STATUS = {"Status":{"Module":1,"FriendlyName":["LED"],"Topic":"LED","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchT
opic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:40:53 MQT: stat/LED/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restar
t","Uptime":"0T00:05:03","StartupUTC":"2018-10-18T19:35:50","Sleep":0,"BootCount":6,"SaveCount":22,"SaveAddress":"F6000"}}

20:40:53 MQT: stat/LED/STATUS2 = {"StatusFWR":{"Version":"6.2.1.17","BuildDateTime":"2018-10-18T21:31:35","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:40:53 MQT: stat/LED/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Balazs","sonoff"],"TelePeriod":300,"SetOption":["00008009","55A18
000","00000000"]}}
20:40:53 MQT: stat/LED/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","240003A0
","23B617CE","00001BC0"]}}
20:40:53 MQT: stat/LED/STATUS5 = {"StatusNET":{"Hostname":"LED-4609","IPAddress":"192.168.1.176","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B
4:E6:2D:15:72:01","Webserver":2,"WifiConfig":5}}
20:40:53 MQT: stat/LED/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.240","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_157201","MqttUser":"DVES_USER","MqttType":1,"M
AX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:40:53 MQT: stat/LED/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 18 19:40:53 2018","Local":"Thu Oct 18 20:40:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2
018","Timezone":1,"Sunrise":"07:15","Sunset":"17:55"}}
20:40:53 MQT: stat/LED/STATUS10 = {"StatusSNS":{"Time":"2018-10-18T20:40:53"}}
20:40:53 MQT: stat/LED/STATUS11 = {"StatusSTS":{"Time":"2018-10-18T20:40:53","Uptime":"0T00:05:03","Vcc":3.014,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Balazs","BSSId":"34:12:98:05:D4:58",
"Channel":1,"RSSI":74}}}

andrethomas2, 2018-10-18:
Closing this issue as it has been answered. Thank you for your valued inputs.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-18 by andrethomas2