arendst/Tasmota #4046: Wemos clone updated and running 6.2.1.15, but COM LED blinking every 2sec?

roguestreak, 2018-10-13:
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Development/Compiler/Upload tools used : Arduino 1.8.5
- [X ] Hardware used : Wemos D1 Mini clone
- [ ] Provide the output of command ``status 0`` :
```
10:47:11 MQT: stat/sonoff_wemos1/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff_Wemos1"],"Topic":"sonoff_wemos1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:21","StartupUTC":"2018-10-13T00:27:50","Sleep":0,"BootCount":64,"SaveCount":86,"SaveAddress":"F4000"}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS2 = {"StatusFWR":{"Version":"6.2.1.15","BuildDateTime":"2018-10-12T12:00:48","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OPTUSFEFF922",""],"TelePeriod":300,"SetOption":["80008009","55818000","00000001"]}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000007C0"]}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_wemos1-5330","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:6E:34:D2","Webserver":2,"WifiConfig":2}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6E34D2","MqttUser":"openhabian","MqttType":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 13 00:47:11 2018","Local":"Sat Oct 13 10:47:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"05:13","Sunset":"17:54"}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS10 = {"StatusSNS":{"Time":"2018-10-13T10:47:11","DHT11":{"Temperature":23.8,"Humidity":69.0},"TempUnit":"C"}}
10:47:11 MQT: stat/sonoff_wemos1/STATUS11 = {"StatusSTS":{"Time":"2018-10-13T10:47:11","Uptime":"0T00:19:21","Vcc":2.710,"POWER":"OFF","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"OPTUSFEFF922","BSSId":"BC:AE:C5:EA:C9:A7","Channel":2,"RSSI":100}}}```

Last night I updated a wemos D1 clone to 6.2.1.15 via OTA, first time stuffed up so reflashed via serial, second OTA attempt was successful.  Everything is running as per prior to flash, but now the COM LED is doing a bright flash every 2 seconds and I can't find why?  I tried setOption31 as it mentioned a 2sec LED flash but this hasn't made any difference.  MQTT is successfully connected and operating correctly, as is the network.

Any ideas, as at night it's quite distracting/annoying, and it wasn't doing this before the update.

Frogmore42, 2018-10-13:
Connect a terminal and see what it's saying?

mike2nl, 2018-10-13:
@roguestreak 
Can you test it with NOT active mqtt, to see it is some that comes from your openhabian?

SetOption3            -> Show current MQTT state
SetOption3 0 / off -> Disable MQTT
SetOption3 1 / on -> Enable MQTT

Frogmore42, 2018-10-13:
If it really is the COM led, that blinks when the serial port has activity.  It sounds like you have serial debug enabled and that is why it is blinking.

mike2nl, 2018-10-13:
@Frogmore42 
right was my 1st thought too. 
About i had to start to build my own mqtt system because i had nothing to test for the VEML6070
and now for VEML6075 which is a new driver.

I have made a lot of mistakes to learn And i saw the COM led blinking too with no debug console on 
it. Strange enough and i can't reproduce that event. 

I had that with the default mqtt client. I had read a discussion between Andre and a another 
user who had mqtt issues to and Andre advised to use a different one and the issues were gone.

So as a i wrote i can't remember the configuration but i'm sure it was without debug com
because my ftdi was dead a half hour before.

roguestreak, 2018-10-13:
status 0 with MQTT disabled:
``16:34:46 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff_Wemos1"],"Topic":"sonoff_wemos1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:34:46 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:45","StartupUTC":"2018-10-13T06:34:01","Sleep":0,"BootCount":69,"SaveCount":92,"SaveAddress":"F6000"}}
16:34:46 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1.15","BuildDateTime":"2018-10-12T12:00:48","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:34:46 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OPTUSFEFF922",""],"TelePeriod":300,"SetOption":["80008001","55818000","00000001"]}}
16:34:46 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000007C0"]}}
16:34:46 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff_wemos1-5330","IPAddress":"192.168.1.28","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:6E:34:D2","Webserver":2,"WifiConfig":2}}
16:34:46 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 13 06:34:46 2018","Local":"Sat Oct 13 16:34:46 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10,"Sunrise":"05:13","Sunset":"17:54"}}
16:34:46 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-10-13T16:34:46","DHT11":{"Temperature":24.3,"Humidity":72.0},"TempUnit":"C"}}
16:34:46 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-10-13T16:34:46","Uptime":"0T00:00:45","Vcc":2.705,"POWER":"OFF","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"OPTUSFEFF922","BSSId":"BC:AE:C5:EA:C9:A7","Channel":5,"RSSI":100}}}``

LED kept blinking the same with it disabled.

Tried SerialLog 0 (was SerialLog 2), no change to the LED.  AFAIK I've never made any specifications about logging, nothing has changed since the firmware update.

arendst, 2018-10-13:
Led is connected to gpio2 on "normal" wemos mini d1. What did you configure on gpio2?

Where is the dht connected to? If not gpio2 try another gpio as your clone may use the current dht gpio for it's led.

roguestreak, 2018-10-13:
Actually it doesn't seem to be a D1 clone, the sellers say that but look nothing like one.  More like a Wemos NodeMC v3?

DHT11 was on GPIO2, WS2812 on GPIO14, nothing else configured and has been this way for about 6 months.

Found a pinout and yes it does appear that GPIO2 (D4) can also be TXD1; I've now moved the DHT11 to GPIO4 (D2) and it looks like the flashing has stopped, so thanks.  Strange that it didn't behave that way in the months it was operating before the upgrade though?  Has there been a driver change and it's polling the sensor a hell of a lot more regularly?  I think I may have noticed the odd flash before, but nothing at all like this.

Clone looks like this one, doesn't have the "module" like the official ones do but I've got a couple of these and they've been working good for some time
https://botland.com.pl/img/art/inne/10908_3.jpg

Pinout:
https://dziadalnfpolx.cloudfront.net/blog/wp-content/uploads/2015/09/esp8266-nodemcu-dev-kit-v3-pins.jpg

andrethomas, 2018-10-13:
I have the same boards as shown in your photo - It just another NodeMCU V3 variant. The LED is connected to D4 GPIO2 - You can confirm this by configuring it as a relay and toggle it.

andrethomas2, 2018-10-13:
Closing this issue as it has been answered. Thank you for your valued inputs.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-13 by andrethomas2