arendst/Tasmota #4047: ESP8266-01 Module

SteveRMann, 2018-10-13:
**Have you look for this feature in other issues and in the wiki?**
Yes
**Is your feature request related to a problem? Please describe.**
No
**Describe the solution you'd like**
Module Type for the ESP8266-01
**Describe alternatives you've considered**
I have my own code, but I also have a number of Sonoff devices, all running Tasmota.  I really like Tasmota and have a few ESP8266-01 modules that I would like to move to Tasmota.  I can flash Tasmota on the device, but there is no Module Type for this device.
**Additional context**
As I said, I wrote my own code on the Arduino IDE for these cheap modules, and after the code is flashed, it has four GPIO ports for use.  As cheap and versatile as these devices are, I can't believe that no one has put Tasmota on them.

mike2nl, 2018-10-13:
@SteveRMann 
Hi Steve,
possible you can start here to read 
-> https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

On that page you have one additional link. I thin it can help you because you are able to write your
own code. Otherwise look at other things in the wiki. A good one..

Frogmore42, 2018-10-13:
There is already generic device.  It exposes all the esp8266 pins for configuration.  Just don't configure the ones you don't have.

Each device configuration takes some amount of flash space and the code size is already quite large.