arendst/Tasmota #4048: Sonoff Basic R2 turns on and off by itself (apparently)

elvenpath, 2018-10-13:
**Describe the bug**
One of my sonoffs just turns on and off by itself, apparently. From what I see in the logs it seems to be doing and on/off every roughly 2 minutes,  I'm starting to believe it might be something related to that specific location or the configuration (sonoff's or ha's) since this is the second device that does this same thing in that location. All my other sonoffs work fine. Here's the console log:

```
07:41:24 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:41:24 MQT: stat/dormitor-mic/POWER = ON
07:41:24 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:41:24 MQT: stat/dormitor-mic/POWER = OFF
07:43:14 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:43:14 MQT: stat/dormitor-mic/POWER = ON
07:43:14 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:43:14 MQT: stat/dormitor-mic/POWER = OFF
07:43:18 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:43:18 MQT: stat/dormitor-mic/POWER = ON
07:43:18 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:43:18 MQT: stat/dormitor-mic/POWER = OFF
07:45:34 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:45:34 MQT: stat/dormitor-mic/POWER = ON
07:45:34 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:45:34 MQT: stat/dormitor-mic/POWER = OFF
07:46:10 MQT: tele/dormitor-mic/STATE = {"Time":"2018-10-13T07:46:10","Uptime":"12T17:50:53","Vcc":3.209,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","RSSI":82,"APMac":"XXXX"}}
07:46:10 MQT: tele/dormitor-mic/SENSOR = {"Time":"2018-10-13T07:46:10","Switch1":"ON"}
07:51:10 MQT: tele/dormitor-mic/STATE = {"Time":"2018-10-13T07:51:10","Uptime":"12T17:55:53","Vcc":3.208,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","RSSI":80,"APMac":"XXXX"}}
07:51:10 MQT: tele/dormitor-mic/SENSOR = {"Time":"2018-10-13T07:51:10","Switch1":"ON"}
07:52:40 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:52:40 MQT: stat/dormitor-mic/POWER = ON
07:52:40 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:52:40 MQT: stat/dormitor-mic/POWER = OFF
07:54:20 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:54:20 MQT: stat/dormitor-mic/POWER = ON
07:54:20 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:54:20 MQT: stat/dormitor-mic/POWER = OFF
07:56:10 MQT: tele/dormitor-mic/STATE = {"Time":"2018-10-13T07:56:10","Uptime":"12T18:00:53","Vcc":3.208,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","RSSI":82,"APMac":"XXXX"}}
07:56:10 MQT: tele/dormitor-mic/SENSOR = {"Time":"2018-10-13T07:56:10","Switch1":"ON"}
07:58:43 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
07:58:43 MQT: stat/dormitor-mic/POWER = ON
07:58:43 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
07:58:43 MQT: stat/dormitor-mic/POWER = OFF
08:01:10 MQT: tele/dormitor-mic/STATE = {"Time":"2018-10-13T08:01:10","Uptime":"12T18:05:53","Vcc":3.208,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXX","RSSI":82,"APMac":"XXXX"}}
08:01:10 MQT: tele/dormitor-mic/SENSOR = {"Time":"2018-10-13T08:01:10","Switch1":"ON"}
08:02:00 MQT: tele/dormitor-mic/UPTIME = {"Time":"2018-10-13T08:02:00","Uptime":"12T18:06:43"}
08:04:36 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
08:04:36 MQT: stat/dormitor-mic/POWER = ON
08:04:36 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
08:04:36 MQT: stat/dormitor-mic/POWER = OFF
08:04:51 MQT: stat/dormitor-mic/RESULT = {"POWER":"ON"}
08:04:51 MQT: stat/dormitor-mic/POWER = ON
08:04:51 MQT: stat/dormitor-mic/RESULT = {"POWER":"OFF"}
08:04:51 MQT: stat/dormitor-mic/POWER = OFF
```
I've tried playing with PowerRetain flag but I didn't notice any difference.

I've checked the issues and wiki, can't  seem to find anything that might help me.
HW: Sonoff Basic R2, V1.0
```
08:08:43 CMD: status 0
08:08:43 MQT: stat/dormitor-mic/STATUS = {"Status":{"Module":1,"FriendlyName":["Dormitor Mic"],"Topic":"dormitor-mic","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:08:43 MQT: stat/dormitor-mic/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"12T18:13:26","StartupUTC":"2018-09-30T11:55:17","Sleep":0,"BootCount":74,"SaveCount":293,"SaveAddress":"F7000"}}
08:08:43 MQT: stat/dormitor-mic/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-16T22:25:24","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:08:43 MQT: stat/dormitor-mic/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":1,"LogHost":"XXXX","LogPort":514,"SSId":["XXXX",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
08:08:43 MQT: stat/dormitor-mic/STATUS4 = {"StatusMEM":{"ProgramSize":459,"Free":544,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","6010A590","0C000008","210217C6","00000000"]}}
08:08:43 MQT: stat/dormitor-mic/STATUS5 = {"StatusNET":{"Hostname":"dormitor-mic","IPAddress":"192.168.1.86","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.3","Mac":"DC:4F:22:45:C7:C4","Webserver":2,"WifiConfig":0}}
08:08:43 MQT: stat/dormitor-mic/STATUS6 = {"StatusMQT":{"MqttHost":"XXXX","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_45C7C4","MqttUser":"XXXX","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:08:43 MQT: stat/dormitor-mic/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 13 06:08:43 2018","Local":"Sat Oct 13 08:08:43 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
08:08:43 MQT: stat/dormitor-mic/STATUS10 = {"StatusSNS":{"Time":"2018-10-13T08:08:43","Switch1":"ON"}}
08:08:43 MQT: stat/dormitor-mic/STATUS11 = {"StatusSTS":{"Time":"2018-10-13T08:08:43","Uptime":"12T18:13:26","Vcc":3.208,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","RSSI":82,"APMac":"XXXX"}}}
```

I can't reproduce the behavior on a different device in a different location.

andrethomas, 2018-10-13:
No thanks for not completing the template as requested.

There is a theory that basic R2's have a power supply issue during startup and the solution which has worked for most people so far is to use the pre-built firmware that was made light-weight for the R2's specifically from here:

http://thehackbox.org/tasmota/

You will see there is a firmware file sonoff-basic_R2.bin which can be used for the R2.

The problem does not present in all sonoff R2's but this has worked for some R2 users.

andrethomas2, 2018-10-13:
Closing this issue as it has been answered as best possible given that the template (when the user created this issue) was either not completed at all or that the information which was completed is insufficient to provide further meaningful responses.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-13 by andrethomas2

elvenpath, 2018-10-13:
Thanks for your reply . Sorry if I've missed anything, I've tried filling in every information I had. If you think I've missed anything I can post it here asap. Is it possible to happen only in a specific location for a specific configuration? Like I wrote above, this is the second device within a week in that location that has this issue, which seems quite suspicious to me. 
I'll update the firmware today.

Thanks.

andrethomas, 2018-10-13:
It might just be that it is two devices from the same manufacturing batch. Report back after flashing sonoff-basic_R2.bin

You may also consider complete erase using esptool or equivalent to ensure that there is no bad data in the SPI flash prior to flashing the firmware.