arendst/Tasmota #4050: APP: Button1 multi-press 1 automatically turns Shelly1 on and off

420Dope, 2018-10-13:
Hi,

i have a Shelly 1 device flashed with Tasmota 6.2.1. The Shelly 1 is installed behind the switch right to the door and turns the light on and off. All works fine, i can turn it on with the switch and off with homebridge.
But from time to time the device goes automatically on and off. One time it goes after maybe 3 mins on or off, other times after may 11 min. Its completly random.
I have check the log and when "APP: Button1 multi-press 1" "SCR: Button" appears, the device turns automatically on or off. 
```
13:20:50 APP: Button1 multi-press 1
13:20:50 SRC: Button
13:20:50 RSL: RESULT = {"POWER":"ON"}
13:20:50 RSL: POWER = ON
13:20:50 CFG: Saved to flash at F8, Count 172, Bytes 3584
13:20:53 WIF: Checking connection...
13:20:53 WIF: Connected
13:21:13 WIF: Checking connection...
13:21:13 WIF: Connected
13:21:30 APP: Button1 multi-press 1
13:21:31 SRC: Button
13:21:31 RSL: RESULT = {"POWER":"OFF"}
13:21:31 RSL: POWER = OFF
```
So the question is, what is "Button1 multi-press 1" and how can i fix it?

ascillato, 2018-10-13:
Hi, please update to latest version. That bug is already fixed. Please, use the precompiled bins from development branch at http://thehackbox.org/tasmota/

closed 2018-10-13 by ascillato2

rcruikshank, 2020-05-20:
I have this problem with Tasmota 8.3.1.1 on a [Medion smart plug.](https://templates.blakadder.com/medion_life_S85225.html) 
Switches off randomly with the following output to console:
`
09:07:23 APP: Button1 multi-press 1
09:07:23 MQT: stat/medion1/BUTTON1T = {"TRIG":"SINGLE"}
09:07:23 SRC: Button
09:07:23 MQT: tele/medion1/STATE = {"Time":"2020-05-21T09:07:23","Uptime":"0T00:34:08","UptimeSec":2048,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Cecil","BSSId":"00:11:32:72:0E:2D","Channel":11,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:18"}}
09:07:23 MQT: stat/medion1/RESULT = {"POWER":"OFF"}
09:07:23 MQT: stat/medion1/POWER = OFF (retained)
09:07:24 CFG: Saved to flash at F6, Count 517, Bytes 4096
`

rokmohar, 2020-06-11:
I have the same problem. Did anybody find any solution yet?

effelle, 2020-06-11:
Since this seems an uncommon problem nowadays, there is a good chance the source of the issue for your device is some kind of noise running through the wires (if is an external button) or traces. 
Hardware side you could add an RC filter, software side you could play a bit with `buttondebounce` to try to limit false positive or avoid the use of first press to toggle the relay (using 2 clicks or a short hold time). Another option is to set it as a switch instead and using a compatible switchmode (like 5 or 6). Obviously you will not have the complete multipress system with a switch.
The latter is what I used with one problematic device: since switches can't have the fully fledged multipress buttons they are less picky and could help. 
In my case later on I re-soldered the pins of the internal button and solve definitely my problem.

rokmohar, 2020-06-12:
Thank you for your quick reply! I forgot to execute `SetOption59 1` on this switch, I think this might have solved the issue, I will have to wait a bit to see. Otherwise I might try some of your suggestions.

rcruikshank, 2020-06-12:
ok 
GPIO3 set to Switch1 from Button1
and
SwitchMode 3 to toggle
Thanks

chris-hatton, 2022-08-04:
The phantom button presses experienced with these Medion Smart switches appear to be caused by electrical design/build quality issues in hardware.  In my view, any attempt to retain the button functionality by filtering its input in software is unacceptable.

My preferred solution is to disable the hardware button completely by ignoring it in Tasmota, by applying this template in configuration:
```
{"NAME":"Medion","GPIO":[0,0,0,0,134,132,0,0,131,56,21,0,0],"FLAG":0,"BASE":52}
```
This is the same as the template found [here](https://templates.blakadder.com/medion_life_S85225.html) but with `GPIO03` (Button1) mapping set to `0` (None), effectively disconnecting it.