arendst/Tasmota #4051: Feature Request - EMS Bus Adapter integration

Chaot1, 2018-10-13:
**Have you look for this feature in other issues and in the wiki?**
yes
**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._
no
**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
using original Versin
**Additional context**
_Add any other context or screenshots about the feature request here._
https://github.com/proddy/EMS-ESP-Boiler
https://github.com/bbqkees/Nefit-Buderus-EMS-bus-Arduino-Domoticz
**(Please, remember to close the issue when the problem has been addressed)**

Is it possible to integrate the Adapter from bbqkees in Tasmota Firmware for reading EMS controled heating systems an transmit those datas per MQTT

Chaot1, 2018-10-16:
Yes, ist a heating Control like described in that brochure.
proddy and bbqkees did a lot of work already on it as you can see on my two links.
But i still hope that anybody can integrate that in Tasmota to be useable on different hardware.

ascillato, 2018-11-12:
Hi,

Is there someone working on a driver for Tasmota for this?

closed 2018-11-14 by ascillato2