arendst/Tasmota #4055: Behavior of .bin upload  with ESPtool and Arduino IDE

upendrawww9, 2018-10-14:
Usecase 1: Pass

Compile the code with Arduino IDE upload the code
Change a parameter like Host name from MQTT or WEB interface. Changes are saved and its effects with changed parameter when reboot

Usecase 2: Fail

Compile the code with Arduino IDE.
upload code/binary with ESPTool. Its working till now
 
Change a parameter like Host name from MQTT or WEB interface. Changes are saved and reboot
**Now the changes are not seen.**

I have checked this more than 10 days.

I want to do with ESPTool

Can you help this behavior of the code.

*IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Frogmore42, 2018-10-14:
You should visit the chat, listed above.

You have provided none of the information necessary to help solve the issue.

closed 2018-10-14 by ascillato2

upendrawww9, 2018-10-14:
Where is the chat info I can login

Frogmore42, 2018-10-14:
Searched the problem in the chat (https://discord.gg/Ks2Kzd4)