arendst/Tasmota #4057: topic command set fails

clifforddw, 2018-10-14:
When setting the "Topic" via a backlog or the console is some cases it fails, reverting to "sonoff" it seems to happen when the topic is the same is a friendly name

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x ] _Searched the problem in issues and in the wiki_
- [x ] _Hardware used_ : sonoff basic
- [c ] _Development/Compiler/Upload tools used_ : web console and termite
- [x ] _Provide the output of command_``status 0`` :
```
16:30:24 MQT: stat/cnelight/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["cnelight14"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:23","StartupUTC":"2018-10-14T14:26:01","Sleep":0,"BootCount":10,"SaveCount":59,"SaveAddress":"F9000"}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["home",""],"TelePeriod":300,"SetOption":["00008029","55818000","00000000"]}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS5 = {"StatusNET":{"Hostname":"cnelight14","IPAddress":"192.168.1.39","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"84:F3:EB:93:B3:87","Webserver":2,"WifiConfig":5}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.21","MqttPort":1883,"MqttClientMask":"cnelight14","MqttClient":"cnelight14","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 14 14:30:24 2018","Local":"Sun Oct 14 16:30:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"08:09","Sunset":"19:03"}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-14T16:30:24","Switch1":"OFF"}}
16:30:24 MQT: stat/cnelight/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-14T16:30:24","Uptime":"0T00:04:23","Vcc":2.988,"POWER":"OFF","Wifi":{"AP":1,"SSId":"4DEither","RSSI":100,"APMac":"82:2A:A8:C4:D9:E0"}}}
```

**To Reproduce**

This is the backlog command used : Backlog SSID1 home; Password1 pssword; GPIO14 09; PowerRetain ON; MqttHost 192.168.1.21; fulltopic %prefix%/cnelight/%topic%/; Hostname cnelight14; MqttClient cnelight14; FriendlyName1 cnelight14; Timezone 2; switchmode1 4

or in the console : 
16:26:34 CMD: Topic cnelight14
16:26:34 MQT: stat/cnelight/sonoff/RESULT = {"Topic":"sonoff"}
16:26:56 CMD: topic
16:26:57 MQT: stat/cnelight/sonoff/RESULT = {"Topic":"sonoff"}
1

**Expected behavior**
Topic should be set to cnelight14

andrethomas, 2018-10-14:
Tried to reproduce...

Before setting, let me check the current topic:
16:45:48 CMD: topic
16:45:48 MQT: stat/sonoff/RESULT = {"Topic":"sonoff"}

Then set the topic:
16:46:11 CMD: Topic cnelight14
16:46:11 MQT: tele/sonoff/LWT =  (retained)
16:46:11 MQT: stat/sonoff/RESULT = {"Topic":"cnelight14"}

The device reports set topic and reboots.
16:46:32 CMD: topic
16:46:32 MQT: stat/cnelight14/RESULT = {"Topic":"cnelight14"}

I'm guessing it has something to do with the other settings you are passing using backlog but not sure.

Also, you can set the topic on the web ui using menu Configuration|Configure MQTT and then you get this screen:

[image]

andrethomas, 2018-10-14:
Best guess it is this
`fulltopic %prefix%/cnelight/%topic%/`
Is there a specific reason why you are inserting a constant?

andrethomas, 2018-10-14:
Strange, because whether from console or command line it stores it to the same setting location in the configuration.

@ascillato Could this be a bug for the use case when a constant is inserted in the topic?

ascillato2, 2018-10-14:
Will investigate

clifforddw, 2018-10-15:
One other point, that makes it really tough is that it does not happen consistently. I programmed 25 devices this way, using the backlog to set them up and only 4 of them had this behavior.

ascillato, 2018-10-16:
Hi,

After several tests I could not reproduce this issue. Sorry.

May be, you have a retained message in your broker that resets your topic ???

ascillato, 2018-10-16:
Please, also try the latest version of the development branch at http://thehackbox.org/tasmota/

andrethomas, 2018-10-16:
One last shot - I see you're on 6.2.1

Can you flash with one of the not working devices with the development binary from http://thehackbox.org/tasmota/ and give us a status 0 along with whether or not the behavior is solved?

clifforddw, 2018-10-16:
@ascillato I don't think it is an mqtt issue as the console command quoted in the initial post came back almost instantly with the fall back to "sonoff"

@andrethomas yea will flash and feedback, happy to help in any way. If there is any other way to debug or trace also happy yo assist.

ascillato, 2018-10-16:
Hi,

Have you managed to solve your issue?

clifforddw, 2018-10-17:
@ascillato, It is not a crisis as the workaround is to use the the MQTT UI configuration so there is no hold up. Mostly filed this to contribute back. 

Will test with the daily build this weekend and revert.

clifforddw, 2018-10-17:
Hi all some more testing. 

Clean sonoff basic, flash with Version 6.2.1-2_3_0

Same backlog string as above and the topic set fails "cnelighttest1" falling back to "sonoff"

I then set the topic using the Serial,Termite to "cnelighttest" so with no integer at the end.

This works and the device resets 

Then I  tried to set  the topic back to "cnelighttest1" and it fails:

Here is a trace from the termite session : [Trace.txt](https://github.com/arendst/Sonoff-Tasmota/files/2488734/Trace.txt)

Based on this I suspect the bug has something to do with a trailing integer

ascillato2, 2018-10-17:
Hi, I test that but with last version. Could you make again your test but with the version from http://thehackbox.org/tasmota/ 

Thanks

ascillato2, 2018-10-17:
Hi, when you have time, please share your results with the last version.

If doesn't work, please, ask to reopen this issue. 

Thanks

closed 2018-10-17 by ascillato2