arendst/Tasmota #4058: Why rule can not survive the restart? (it become inactive)

easyyu, 2018-10-14:
I serched on google and nowhere could not find the solution.
I have one simple rule for example:
`Rule1 on AM2301-14#Temperature=nan do Power1 0 endon`
If I activate this rule with command:
`rule1 1`
I got the message:
`15:42:55 MQT: stat/sonoff-Rome/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":464,"Rules":"on AM2301-14#Temperature=nan do restart 1 endon"}`
and the rule is active, till next restart.
After the restart witho command `rule1` I got message:
`15:44:41 MQT: stat/sonoff-Rome/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":464,"Rules":"on AM2301-14#Temperature=nan do restart 1 endon"}`
Rule is OFF.
Why it can not survive the restart any rules?
I just updated firmware to latest one 6.2.1 from webinterface.

Here is output from staus 0:
`15:38:22 CMD: status 0
15:38:23 MQT: stat/sonoff-Rome/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff-Rome","ButtonTopic":"sonoff","Power":1,"PowerOnState":1,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:38:23 MQT: stat/sonoff-Rome/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:45","StartupUTC":"2018-10-14T14:37:38","Sleep":0,"BootCount":172,"SaveCount":538,"SaveAddress":"F9000"}}
15:38:23 MQT: stat/sonoff-Rome/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:38:23 MQT: stat/sonoff-Rome/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["mywifi",""],"TelePeriod":10,"SetOption":["00008009","55818000","00000001"]}}
15:38:23 MQT: stat/sonoff-Rome/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
15:38:23 MQT: stat/sonoff-Rome/STATUS5 = {"StatusNET":{"Hostname":"sonoff-Rome-6044","IPAddress":"192.168.1.134","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"5C:CF:7F:7A:37:9C","Webserver":2,"WifiConfig":2}}
15:38:23 MQT: stat/sonoff-Rome/STATUS6 = {"StatusMQT":{"MqttHost":"fifieasy.almostmy.com","MqttPort":31883,"MqttClientMask":"Rome","MqttClient":"Rome","MqttUser":"easydoor","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:38:23 MQT: stat/sonoff-Rome/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 14 14:38:23 2018","Local":"Sun Oct 14 15:38:23 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:09","Sunset":"18:03"}}
15:38:23 MQT: stat/sonoff-Rome/STATUS10 = {"StatusSNS":{"Time":"2018-10-14T15:38:23","AM2301-03":{"Temperature":26.3,"Humidity":99.9},"AM2301-14":{"Temperature":29.0,"Humidity":40.9},"TempUnit":"C"}}
15:38:23 MQT: stat/sonoff-Rome/STATUS11 = {"StatusSTS":{"Time":"2018-10-14T15:38:23","Uptime":"0T00:00:45","Vcc":3.224,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":54,"APMac":"78:29:ED:4B:0D:49"}}}
15:38:24 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T15:38:24","Uptime":"0T00:00:46","Vcc":3.226,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":58,"APMac":"78:29:ED:4B:0D:49"}}`

andrethomas, 2018-10-14:
Hi

Thanks for opening an issue.

Let me try to reproduce with a cleanly flashed wemos board

I set the rule with

`Rule1 on AM2301-14#Temperature=nan do Power1 0 endon`

and get response:
```
16:59:35 CMD: Rule1 on AM2301-14#Temperature=nan do Power1 0 endon
16:59:35 MQT: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":465,"Rules":"on AM2301-14#Temperature=nan do Power1 0 endon"}
```

Turn on rule with:
```
rule1 1
```
and get response
```
17:00:56 CMD: rule1 1
17:00:56 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":465,"Rules":"on AM2301-14#Temperature=nan do Power1 0 endon"}
```
So just to check that it is in fact on I do command:
```
rule
```
Which gives me output as follows, confirming that it is enabled
```
17:01:48 CMD: rule
17:01:48 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":465,"Rules":"on AM2301-14#Temperature=nan do Power1 0 endon"}
```
Now for a reboot fron the web ui button, I get the expected reboot serial via console:
```
17:02:38 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_102FD4, GroupTopic sonoffs) Version 6.2.1.15-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-4052...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-4052 with IP address 192.168.42.42
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff/POWER = 
00:00:05 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1.15","FallbackTopic":"DVES_102FD4","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-4052","IPAddress":"192.168.42.42"}
00:00:05 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/sonoff/POWER = OFF
17:02:48 MQT: tele/sonoff/UPTIME = {"Time":"2018-10-14T17:02:47","Uptime":"0T00:00:09"}
```
Now I give the following command to check what is in rules;
```
rule
```
And get the output, as expected the rule is still very much active.
```
17:04:06 CMD: rule
17:04:06 MQT: stat/sonoff/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":465,"Rules":"on AM2301-14#Temperature=nan do Power1 0 endon"}
```

Sorry, but I cannot seem to reproduce what you are experiencing.

The only difference is that I am running the current development version 6.2.1.15 but I am not aware that there existed a bug in rules in any recent history.

ascillato, 2018-10-14:
If after a restart the rules goes off, that means that there is something wrong in your config that makes Tasmota hungs after restart. So, Tasmota on each hung will disables some features.

For better troubleshooting, will be better to enable full log using command weblog 4. Please, set your rules on again and try to catch the console output after a restart. May be better to catch that using the serial output.

Also try latest version from development branch from http://thehackbox.org/tasmota/

easyyu, 2018-10-14:
Strange..
I hav nothing extra in settings..just one active (or I would like to be active) roule and active mqtt to my mosquitto server.
I do not see nothing strange in log.
The problem is that sonoff Basic is not phihicly accessable now it say far away from me.
What can I do more to solv this problem?

Here is the log with weblog 4 before restart:
`17:25:39 WIF: Connected
17:25:46 CMD: rule1 1
17:25:46 SRC: WebConsole from xx.xx.xx.xx
17:25:46 RSL: Received Topic /rule1, Data Size 1, Data 1
17:25:46 RSL: Group 0, Index 1, Command RULE, Data 1
17:25:46 MQT: stat/sonoff-Rome/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":464,"Rules":"on AM2301-14#Temperature=nan do restart 1 endon"}
17:25:46 CFG: Saved to flash at FA, Count 545, Bytes 3584
17:25:48 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:25:48","Uptime":"0T01:41:17","Vcc":3.226,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":52,"APMac":"78:29:ED:4B:0D:49"}}
17:25:48 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:25:48","AM2301-03":{"Temperature":25.3,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.1},"TempUnit":"C"}
17:25:58 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:25:58","Uptime":"0T01:41:27","Vcc":3.223,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":48,"APMac":"78:29:ED:4B:0D:49"}}
17:25:58 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:25:58","AM2301-03":{"Temperature":25.4,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.2},"TempUnit":"C"}
17:25:59 WIF: Checking connection...
17:25:59 WIF: Connected
17:25:59 CMD: restart 1
17:25:59 SRC: WebConsole from xx.xx.xx.xx
17:25:59 RSL: Received Topic /restart, Data Size 1, Data 1
17:25:59 RSL: Group 0, Index 1, Command RESTART, Data 1
17:25:59 MQT: stat/sonoff-Rome/RESULT = {"Restart":"Restarting"}`

and after restart:
`00:00:00 CFG: Loaded from flash at F7, Count 548
00:00:00 APP: Some settings have been reset (4)
00:00:00 APP: Boot Count 180
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff (Topic sonoff-Rome, Fallback Rome, GroupTopic sonoffs) Version 6.2.1-2_3_0
00:00:00 CFG: Saved to flash at F6, Count 549, Bytes 3584
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 mywifi in mode 11N as sonoff-Rome-6044...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-Rome-6044.local with IP address 192.168.1.134
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff-Rome/LWT = Online (retained)
00:00:06 MQT: cmnd/sonoff-Rome/POWER = 
00:00:06 MQT: Subscribe to cmnd/sonoff-Rome/#
00:00:06 MQT: Subscribe to cmnd/sonoffs/#
00:00:06 MQT: Subscribe to cmnd/Rome/#
00:00:06 MQT: tele/sonoff-Rome/INFO1 = {"Module":"Sonoff Basic","Version":"6.2.1","FallbackTopic":"Rome","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/sonoff-Rome/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-Rome-6044","IPAddress":"192.168.1.134"}
00:00:06 MQT: tele/sonoff-Rome/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/sonoff-Rome/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff-Rome/POWER = ON
00:00:06 APP: (UTC) Sun Oct 14 16:26:15 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
17:26:23 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:26:23","Uptime":"0T00:00:14","Vcc":3.222,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":50,"APMac":"78:29:ED:4B:0D:49"}}
17:26:23 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:26:23","AM2301-03":{"Temperature":25.3,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.1},"TempUnit":"C"}
17:26:33 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:26:33","Uptime":"0T00:00:24","Vcc":3.225,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":54,"APMac":"78:29:ED:4B:0D:49"}}
17:26:33 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:26:33","AM2301-03":{"Temperature":25.4,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.3},"TempUnit":"C"}
17:26:34 WIF: Checking connection...
17:26:34 WIF: Connected
17:26:43 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:26:43","Uptime":"0T00:00:34","Vcc":3.225,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":46,"APMac":"78:29:ED:4B:0D:49"}}
17:26:43 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:26:43","AM2301-03":{"Temperature":25.3,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.3},"TempUnit":"C"}
17:26:45 CMD: rule1
17:26:45 SRC: WebConsole from xx.xx.xx.xx
17:26:45 RSL: Received Topic /rule1, Data Size 0, Data 
17:26:45 RSL: Group 0, Index 1, Command RULE, Data 
17:26:45 MQT: stat/sonoff-Rome/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":464,"Rules":"on AM2301-14#Temperature=nan do restart 1 endon"}
17:26:53 MQT: tele/sonoff-Rome/STATE = {"Time":"2018-10-14T17:26:53","Uptime":"0T00:00:44","Vcc":3.223,"POWER":"ON","Wifi":{"AP":1,"SSId":"mywifi","RSSI":46,"APMac":"78:29:ED:4B:0D:49"}}
17:26:53 MQT: tele/sonoff-Rome/SENSOR = {"Time":"2018-10-14T17:26:53","AM2301-03":{"Temperature":25.4,"Humidity":99.9},"AM2301-14":{"Temperature":23.0,"Humidity":76.4},"TempUnit":"C"}
17:26:54 WIF: Checking connection...
17:26:54 WIF: Connected`

easyyu, 2018-10-14:
Only this part seams to be strange from log on the begoning of restart:
`00:00:00 CFG: Loaded from flash at F7, Count 548
00:00:00 APP: Some settings have been reset (4)
00:00:00 APP: Boot Count 180
00:00:00 SRC: Restart`

But there is no explaination why.
Help?

andrethomas, 2018-10-14:
Other than what @ascillato already said I cannot think of any other reason why it is behaving in that way other than that there is possibly bad data in the SPI flash.

A reset to default settings may solve this but it would depend on whether or not you will be able to connect to it in AP mode to get it reconnected to your wifi.

easyyu, 2018-10-14:
yes..that is too risky if I ca not access it anymore...

andrethomas, 2018-10-14:
Some sonoff basic's have borderline power supplies so we have observed that this may happen in some cases to sonoff basic's - specifically from users who have sonoff basic R2's although I am not sure how they determine that they have a R2 as mine were not marked in any specific way.

easyyu, 2018-10-14:
but..ok, let it say that it is chepp the HW and all other and to do not want to write the command after reboot once twice..but never?

andrethomas, 2018-10-14:
Well, the problem cannot be reproduced on a normally functional device - so it's not a Tasmota firmware issue and Tasmota is not responsible for the quality control of the hardware you are using.

I think we've exhausted all options except the ones which you do not want to do because the device is in a hard to reach location.

closed 2018-10-14 by ascillato2

easyyu, 2018-10-14:
I didnt tell that I do not wanna, just that next few weeks it is unreachable physicly to me.
But of course that I will try your suggestion!

andrethomas, 2018-10-14:
Sure, understood.

Perhaps just a shot in the dark, instead of using rule1 have you tried rule2 ?
Rule2 on AM2301-14#Temperature=nan do Power1 0 endon

easyyu, 2018-10-15:
Hah..thx for idea..
Unbelivable but with rule > than 1 it survive the restart.
But why?

ascillato, 2018-10-15:
Because you have a problem in the flash, so there is something that makes Tasmota to hung at boot time, so, Tasmota have a failsafe that starts to disables functions to make it boot. So, after it hungs 4 times in a row (less than 4 seconds between reboots) , it disables several things including rule1. Then if hungs again will be disabled rule2 and other features.

That is why. So, your issue is in the flash. May be a wrong value in some part.

To solve that, you will need to erase all flash with esptool.py and flash again. The recommend are the precompiled bins from development branch.

easyyu, 2018-10-15:
Understand how it work..
I downloaded as zip this project and serach in it the esptool.py but I coudn't found.
Where to find and how to start?
Is there any instruction?
U mean to do by myself compile from: https://github.com/arendst/Sonoff-Tasmota
?
Thx for ideas!

andrethomas, 2018-10-15:
esptool.py comes from mars...

andrethomas, 2018-10-15:
You really need to stop cluttering the github issues with questions that are already answered by hard work from community members - https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

andrethomas2, 2018-10-15:
This issue is being closed and locked because a workaround has been given as an interim solution until the user can find the time when the device can be accessed physically to resolve what appears to be a hardware malfunction.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.