arendst/Tasmota #4059: Sonoff POW R2, is it possible to change energy values when switch is toggled?

Klagio, 2018-10-14:
Have the below configuration in Home Assistant. All works fine, but the energy (voltage, etc..) are not updated correctly.
For example when I turn the switch OFF all the values should be updated immediately (Voltage to zero, and so on, while the voltage stays at 220 for long time before updating)

I hope I made myself clear

```
switch:
  - platform: mqtt
    name: "Oven power plug"
    state_topic: "stat/oven/POWER"
    command_topic: "cmnd/oven/POWER"
    availability_topic: "tele/oven/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
sensor:
  - platform: mqtt
    name: Oven Current
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Current }}"
    unit_of_measurement: 'amps'
  - platform: mqtt
    name: Oven Power
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Power }}"
    unit_of_measurement: 'watts'
  - platform: mqtt
    name: Oven Voltage
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Voltage }}"
    unit_of_measurement: 'volts'
  - platform: mqtt
    name: Total energy consumed
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Total }}"
    unit_of_measurement: 'kWh'
  - platform: mqtt
    name: Yesterday energy consumed
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Yesterday }}"
    unit_of_measurement: 'kWh'
  - platform: mqtt
    name: Today energy consumed
    state_topic: "tele/oven/SENSOR"
    value_template: "{{ value_json.ENERGY.Today }}"
    unit_of_measurement: 'kWh'
```

Klagio, 2018-10-14:
Now I red that is possible to change from 5 minuites to 10 seconds with TelePeriod 10. Which is fine for reading (although I would prefer 1-2 seconds, is not possible?)
In any case is important to have an instant reading  when there is a toggle (when I switch OFF it should bring sensor values to zero immediately and not after 10 seconds)

andrethomas, 2018-10-14:
Lower than 10 seconds is not supported by default, but you are welcome to compile your own firmware and modify where necessary the value handling of the Settings.tele_period variable.

andrethomas2, 2018-10-14:
Closing this issue as it has been answered. Thank you for your valued inputs.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-14 by andrethomas2

andrethomas, 2018-10-14:
Actually, a work around for this could be that when you get the power off notification on your HA is to send a MQTT back to the device with the payload "STATUS 10"

This will give the following result back to HA via MQTT
`20:37:51 MQT: stat/sound1/STATUS10 = {"StatusSNS":{"Time":"2018-10-14T20:37:51","ENERGY":{"Total":2.300,"Yesterday":0.672,"Today":0.628,"Power":96,"ApparentPower":127,"ReactivePower":83,"Factor":0.76,"Voltage":222,"Current":0.571}}}`

Klagio, 2018-10-14:
Which topic? I tried the below and is not working, any hint?

  - alias: oven to on update status
    trigger:
      - platform: state
        entity_id: switch.oven_power_plug
        to: 'on'
    action:
      - service: mqtt.publish
        data:
          topic: "cmnd/oven/status"
          payload: "STATUS 10"

Klagio, 2018-10-14:
Not working, it takes always around 10 seconds to be updated

Klagio, 2018-10-15:
topic: "cmnd/oven/status"
payload: "10"

problem with the above I receive the opposite value: for example the switch is ON so Voltage is 220, when I turn OFF the switch and send the payload as above, I receive in return Voltage 220, although clearly the Voltage went to 0.
Basically like the above I get the opposite value (zero when ON and a value when OFF)