arendst/Tasmota #4060: Switch2 toggles Relay1 when MQTT broker is stopped or otherwise inaccessible.

jziolkowski, 2018-10-14:
Hello!

I've noticed this repeatable behavior when testing how will my device behave when it loses MQTT connectivity. When the MQTT broker is up and functional, the device works as expected. Button1 toggles relay1. Switch2 sends a MQTT message to either default topic or a custom one (as it is supposed to do in my setup; it's used with HomeAssistant to toggle server-side automation). When I disconnect MQTT broker, button1 still works, but switch2 starts to toggle relay1. As soon as MQTT connectivity is restored, the behavior returns to normal.

It's a Wemos D1 mini, configured as 18 generic. 

D3 -> switch2
D2 -> button1
D1 -> relay1

there's no relay2. There are no rules in place. 

Status 0:

17:42:06 SRC: WebConsole from 10.0.11.1
17:42:06 RSL: Group 0, Index 1, Command STATUS, Data 0
17:42:06 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Lampa TV P"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:42:06 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:06:34","StartupUTC":"2018-10-14T16:35:32","Sleep":0,"BootCount":41,"SaveCount":334,"SaveAddress":"F6000"}}
17:42:06 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:42:06 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Kaszanka",""],"TelePeriod":300,"SetOption":["400AA00B","55818000","00000000"]}}
17:42:07 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
17:42:07 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4791","IPAddress":"10.0.11.125","Gateway":"10.0.11.254","Subnetmask":"255.255.255.0","DNSServer":"10.0.11.249","Mac":"B4:E6:2D:37:72:B7","Webserver":2,"WifiConfig":1}}
17:42:07 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.11.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3772B7","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:42:07 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 14 16:42:07 2018","Local":"Sun Oct 14 17:42:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:09","Sunset":"18:03"}}
17:42:07 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-14T17:42:07","Switch2":"ON"}}
17:42:07 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-14T17:42:07","Uptime":"0T00:06:35","Vcc":2.717,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Kaszanka","RSSI":100,"APMac":"D4:CA:6D:E3:85:2D"}}}

closed 2018-10-14 by ascillato2

jziolkowski, 2018-10-14:
a designed behavior to have switch2 toggle relay1 when mqtt is not available?

ascillato, 2018-10-14:
Yes, because switchtopic is intended for switches. So, when you loose mqtt you will still be able to control your relay.

Because, in some other cases, a different behaviour is needed, you must use rules to achieve the behaviour you want.

It is all explained in the examples in rules at the wiki

jziolkowski, 2018-10-14:
I get it, but: switch2 triggers relay1. It'd make sense with 1->1, 2->2. Why all of a sudden 2->1? What if there would be more switches and/or relays?

ascillato, 2018-10-15:
Sorry, if you don't have relay2 defined means that you have 2 switches to turn relay 1, like a stair light.

Please, for the behaviour you want, was designed the rules as explained in the wiki with examples.

jensuffhaus, 2020-01-25:
@ascillato Firstly, sorry to awaken this old thread ... I found it and it is exactly my problem but I have some more information ...

I understand and agree with the "All Switches" > "Relay 1" if MQTT is not set or unavailable. Good backup.

However ... It seems even with MQTT setup and available this behaviour persists when switchtopic is set to 0. That I think is wrong. If I have switchtopic on 0 with MQTT working then switch2 should not trigger relay1, but it does.

ascillato, 2020-01-25:
If you don't have relay2 defined, all switches will toggle relay1