arendst/Tasmota #4061: Checkbox to keep WiFi option when doing a complete setting reset via web panel

jziolkowski, 2018-10-14:
I think it's self-explanatory.

Having to switch wifi, input WiFi credentials each time you need to reset the device to default settings - while not happening that often perhaps - is tedious and unneeded, because these two options doesn't impact anything serious. I'm aware the webpanel is to be kept minimally complex and feature-rich due to memory constraints, but I think this is a minor change which can save a lot of time overall.

andrethomas, 2018-10-14:
Not a bad idea although I currently use the console when doing a reset to defaults as follows:
`backlog reset 1;ssid1 wifiname;password1 wifikey;`

Just tested it to make sure and I get the following output on the serial console
```
19:04:53 CMD: backlog reset 1;ssid1 Wireless;password1 wifipassword;
19:04:54 MQT: stat/sonoff/RESULT = {"Reset":"Reset and Restarting"}
19:04:54 MQT: stat/sonoff/RESULT = {"SSId1":"Wireless"}
19:04:54 MQT: stat/sonoff/RESULT = {"Password1":"wifipassword"}
19:04:56 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v60000608
~ld

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_102FD4, GroupTopic sonoffs) Version 6.2.1.15-2_3_0
00:00:00 WIF: Connecting to AP1 Wireless in mode 11N as sonoff-4052...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-4052.local with IP address 192.168.42.42

```

andrethomas, 2018-10-14:
Hmm, nevermind that - I forget that I load pre-compiled bins from local web server that already has my wifi configured so it makes no difference at all

andrethomas, 2018-10-14:
`One can dream of a custom build toolchain NodeMCU`

Working on that

andrethomas, 2018-10-14:
Proposed PR https://github.com/arendst/Sonoff-Tasmota/pull/4062

andrethomas2, 2018-10-14:
Closing this issue as it the PR made for this request has been merged into development.

closed 2018-10-14 by andrethomas2