arendst/Tasmota #4063: Sonoff Backlog automatic abort on button/interaction

vontrapp, 2018-10-14:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Problem: I configure a backlog to turn off a light after some delay
Before that delay expires (still in the backlog) a user manually overrides the switch.
The backlog delay expires and the action is performed anyway

**Describe the solution you'd like**
I think a good solution would be a backlog coniguration item that specifies whether or not to abort the backlog if another event/command happens.

It could be aborted if a button is pressed, or if another mqtt command is received, or both, or neither.

**Describe alternatives you've considered**
Rules. It's another learning curve but I believe rules could accomplish the desired result.

**Additional context**
One example is holding the button to activate a timer (implemented with backlog) to turn off the light after a predetermined time. Hold the button, then get in bed and settled in. Light then turns off.

But if the light is manually turned off, then turned back on (left the room but came back) the timer should be cancelled, otherwise the light turns off shortly after you came back into the room.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-14:
Hi,

Sorry but as the complexity for explaining that to new users is high, it is not likely to be implemented.

As a workaround for you to have this behaviour, is set a rule to verify a button press and to not perform the ongoing actions. 

You can set a variable to allow the action and then, the variable can be checked to perform or not your actions. If you press your button, you can modify your variable and your actions will not be performed. 

Let me know if you need help setting this rule.

closed 2018-10-14 by ascillato2

vontrapp, 2018-10-14:
I was actually thinking of using something like:
rule on <buttonpress> do backlog endon

Meaning that the 'backlog' command will abort the backlog. Does this seem like it would work? I'll be trying it sometime soon in any case.

ascillato, 2018-10-14:
Sorry, that will not work.

vontrapp, 2018-10-14:
Here's my console, it seems to work, though not with the button (probably what you were thinking) but instead with a change in power state, which is good enough for me.
```
22:13:52 CMD: rule1 on Power2#State do backlog endon
22:13:52 MQT: stat/bed1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":479,"Rules":"on Power2#State do backlog endon"}
22:13:59 MQT: stat/bed1/RESULT = {"POWER2":"ON"}
22:13:59 MQT: stat/bed1/POWER2 = ON
22:13:59 RUL: POWER2#STATE performs "backlog"
22:13:59 MQT: stat/bed1/RESULT = {"Backlog":"Empty"}
22:14:07 MQT: cmnd/bed1button/POWER2 = HOLD
22:14:08 MQT: stat/bed1/RESULT = {"Delay":600}
22:14:10 MQT: cmnd/bed1button/POWER2 = HOLD
22:14:11 MQT: stat/bed1/RESULT = {"POWER2":"OFF"}
22:14:11 MQT: stat/bed1/POWER2 = OFF
22:14:11 RUL: POWER2#STATE performs "backlog"
22:14:11 MQT: stat/bed1/RESULT = {"Backlog":"Aborted"}
22:14:13 MQT: stat/bed1/RESULT = {"POWER2":"ON"}
22:14:13 MQT: stat/bed1/POWER2 = ON
22:14:13 RUL: POWER2#STATE performs "backlog"
22:14:13 MQT: stat/bed1/RESULT = {"Backlog":"Empty"}

```

vontrapp, 2018-10-14:
It sure would be nice if I could completely decouple the relay control from ALL button variants (toggle/hold/'regular')

ascillato, 2018-10-15:
Yes, you can. For that you need to use rules. See the examples provided in rules at the wiki. Thanks

vontrapp, 2018-10-15:
It doesn't appear that I can.

Example 12 says if switchmode isn't set to 1, then I don't get switch#state=1 ('regular'), however I only get state=2 (toggle) and state=3 (hold) if I have switchmode set to 5 or 6.

How do I get all 3 available for rules at the same time?