arendst/Tasmota #4067: DS18B20 measures its own temperature?

dlemper, 2018-10-15:
Hi everybody,

I‘m using this firmware now on at least 5 devices and like it very much, thank you!

However two of my devices are Wemos D1 Mini with DS18B20 shields and I first wrote my own arduino sketch, that retrieved the current temperature every one second and transfers the value via MQTT. This sketch worked perfectly fine (but doesn‘t have update features or webinterface like tasmota) and after some months I replaced it with tasmota.
After the switch to tasmota, I got higher temperatures, e.g. I felt cold and the DS18B20 told me, it were 25.6 degrees Celsius but in reality it was about 18 degrees.

Then I compared the algorithms of my sketch with the one of tasmota driver for DS18B20, but both result in the same values converted from raw.

Then I reuploaded my sketch today and was irritated, that the temperature value dropped from second to second by around 0.3 degrees.

For me, it looks like the DS18B20 gets warm for unknown reasons (so far) and measures it‘s own temperature, which is higher then room temperature.

Does anybody have the same effect and maybe a solution for this?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dlemper, 2018-10-15:
I forgot to mention, that I configured tasmota to retrieve a new temperature value every 10 seconds, sorry

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Jason2866, 2018-10-15:
Can you try a different 18B20?

dlemper, 2018-10-15:
I already tried both of my wemos boards with ds18b20 shields, same effect..

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

edelstahlratte, 2018-10-15:
Do you have the sensor in very close proximity of the Wemos D1? The ESP gets quite warm...
I have 10 DS18B20 running without problems and reporting plausible temperatures... Some together with a BME280 to cross-check the temperatures...

Frogmore42, 2018-10-15:
The DS18B20 sensors are not known for self-healing.  10s is generally way too fast for room temperature, ie rooms don't change temperature that quickly.  But, that shouldn't cause an issue.  With the addition of rules, I believe the temperature is checked every second.  If the sensor is right near the esp8266, it will be impacted by the ~1W it gives off as heat.

dlemper, 2018-10-15:
@edelstahlratte it looks exactly like the left part of the picture: https://goo.gl/images/sbcAU4
The ds18b20 is stacked upon the wemos.

@Frogmore42 good suggestion, but what irritates me is, that the simple arduino sketch i almost copied together works perfectly fine.. I will put that sketch on pastebin later and put a link here

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

edelstahlratte, 2018-10-15:
I had to put my Sensors at least 10cm away from the ESP8266 because of false (to high) temperature readings (esp. with the BME280 )... You can check this easily by blowing air over the sensor with a small computer fan... The temperature should drop to the expected level...

dlemper, 2018-10-15:
@edelstahlratte thats the easiest solution and i‘ll try that out, thank you! However I don‘t understand, why tasmota needs more power than the arduino sketch...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Jason2866, 2018-10-15:
@dlemper 
Tasmota does even more just reading a sensor🤔

Frogmore42, 2018-10-15:
If you search the issues, you will see some testing of the power utilization of the esp8266.  It is about a watt under normal circumstances.  In the close configuration you have that will impact the temperature reading.  You also have it above the source of heat, which makes it worse.

As others have said, your program is doing less, probably sleeping most of the time.  You can try sleep with Tasmota, there is a setting for that.  It can drop the power consumption significantly and that will lower the impact of the heating.

Jason2866, 2018-10-15:
Please close this issue your question has been answered.
THX

andrethomas2, 2018-10-15:
Closing this issue as it has been answered. 

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-15 by andrethomas2

dlemper, 2018-10-15:
Oh, didn‘t know about the sleep feature and will try that out, thanks everybody!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>