arendst/Tasmota #4068: RF devices

pgext, 2018-10-15:
Hi,
recently I bought a few Sonoff devices. I've run eWeLink app yesterday and I'm disappointed with it :-(
1) I cannot turn on switch for less than 1 minute. 
2) Timers are run no for specified time amount but on whole RTM minutes. I.e. switch on from PIR on 11:45:55 will be off on 11:46:00 not on 11:46:55
3) I got TH16 but there is no option to store/view historical values, graphs
4) I cannot get any notifications via mail, sms

I found Tasmota, Domoticz on Raspberry and it looks fine. But I got questions about RF devices. I saw RF bridge firmware, so it should works correclty. What about Sonoff PIR2 and Door sensor DW1 ? Will it works correclty with Tasmota? 
I saw just https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR501-PIR-Motion-Sensor but this is not Sonoff PIR2 sensor...

Thanks,
Pawel

andrethomas, 2018-10-15:
The Sonoff Bridge and the PIR2 and DW1 sensors are quite widely used and work perfectly with Tasmota.

There are also people who have successfuly used other devices that provide logic output - If you search the issues you will find issues like https://github.com/arendst/Sonoff-Tasmota/issues/3147 that was previously discussed.

andrethomas2, 2018-10-15:
Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-15 by andrethomas2