arendst/Tasmota #4069: Control ifan02 on the basis of a humidity sensor

danielpondruff, 2018-10-15:
Hi,

I have the following question for you.

I want to use the ifan02 to control my fan in the bathroom. Is it possible to run some kind of script with Tasmota?

I want to control the fan level 1-2-3 according to the current humidity.

 Is this possible with Tasmota, or do I have to use EasyESP for this?

Greetings
 Daniel

wongnam, 2018-10-15:
Use Rule to do so. refer to Wiki.

andrethomas2, 2018-10-15:
As indicated by @wongnam please read the Wiki on Rules here https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

Closing this issue as it has been answered.

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-15 by andrethomas2

danielpondruff, 2018-10-15:
Perfect. Thank you 👍