arendst/Tasmota #4071: Device not found

nematia, 2018-10-15:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Development/Compiler/Upload tools used :
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

Device not found

JSON ERROR => 4: Syntax error
Please copy the whole error message and copy it in a new issue on GitHub. Somehow the Tasmota Firmware answered a wrong formatted JSON string which needs to get adjusted in TasmoAdmin.
JSON Answer => { margin-left:10%; margin-right:10%; color:black; background:white; font-family:Verdana, sans-serif;} h2,h3,h4 { font-family:Garamond, Serif; letter-spacing:5px; } span.little { font-size:60%; margin-left:5px} span.red { color:red; } span.green { color:green;}h1 { font-family:Verdana, sans-serif; letter-spacing:10px; margin-bottom:10px;} table.keypad {background:black; color:white; } h4 { color:red;} td.hdl {text-align:left; padding:0px;} td.hdr {text-align:right; padding:0px;}td { font-size:90%; text-align:center; padding:2px;} td.open {background:red; color:black; font-weight:bolder; border-style:outset; } th {text-align:left; background:#00006B; color:white; font-weight:bold;} th#nl {font-size:90%;} th#nr {text-align:right; font-size:90%; padding-right:10px;} td.sp {height:2em;} #content {margin: 5px 5px 5px 5px;} input {font-size:80%; padding:2px} a {text-decoration:none;} a:link {color:white;} a:visited {color:white;} a:hover {color:gray} a:active {color:yellow} .black a:link {color:black} .black a:visited {color:black} div.log {BORDER: #ccc 1px solid; BACKGROUND:#eee; width:300px; text-align:left; padding:3px} div.admin, div.blank { width:700px;} div.admin, div.online {BORDER: #ccc 1px dashed; BACKGROUND:#eeeeee; padding:3px}

**(Please, remember to close the issue when the problem has been addressed)**

andrethomas, 2018-10-15:
Which device was not found?

Also, please update your initial post to include the output of the command

`status 0`

Thanks

nematia, 2018-10-15:
The device is a Sonoff basic running Tasmota firmware 6.2.1. I was getting json syntax error, posted above,  while trying to add this device using Autoscan or when I tried to add it without Autoscan . However, after a couple or tries, I'm now able to add the device. 

I'll go ahead and close this issue. Thank you for your prompt response.