arendst/Tasmota #4077: Putting Sonoff RF R2 in flash mode

pedro1977, 2018-10-16:
I have an R2 version of the Sonoff RF and can’t get it to go into flash ready mode to accept Tasmota. The board seems to have a different layout than those on the wiki page and the RF board is attached horizontally not vertically like the original. Haven’t had any luck with just holding down the button on powering. Any help would be appreciated. Thanks

ascillato, 2018-10-16:
Hi,

You need to push the button before powering up.

pedro1977, 2018-10-16:
Thanks for the quick reply. I’ll try again. So no separate bridging of GPIO1 to ground like the original RF?
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF

SteveDinn, 2018-11-30:
I have a "R2 V1.0" labelled "2017-11-23", and I can't get this thing to go into flash mode at all. I've flipped the switch to 'off', held the button on the side while connecting the serial cables, but it does NOT let me flash it. I've done several Sonoff basics, but the RF Bridge is stubborn!

I'd love to hear if anyone has managed to flash this particular model.