arendst/Tasmota #4081: how to work with WEMOS D1 MINI

rt400, 2018-10-16:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : ARDUINO
- [X] Hardware used : WEMOS D1  MINI
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
HI ,

i use tasmota with WEMOS D1 MINI that connect 6 Relays arduino boards.
now i want to have a button that turn on or off one of the relay .
i try to config some gpios like button or switch but he does not work and make hard reset for the board.

what the diffrent between switch and botton .
i want a 2 switch that turn the relay . now if i turn on from one switch and try to turn on from second switch the relay still be off .

hope that its clear ( my english not that well...)
thanks
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-10-16:
This could be done easy with a TM1638. It is full supported from Tasmota
[image]

rt400, 2018-10-16:
i need only on switch for one relay in the 6 relays
but i want 2 switch that can control together  to relay

Jason2866, 2018-10-16:
Can you do the wiring for the 2 switches if you want to or is the wiring given?
In general this is the wrong place for questions to individual setup.
As the name say it is for Issues related to the software.
Please visit Discord Chat for support under https://discord.gg/Ks2Kzd4
A good place to start is the wiki too :-)

closed 2018-10-16 by ascillato2