arendst/Tasmota #4086: Different over 6°C with SHT30 to BME280

dr-apple, 2018-10-16:
Ich have testet the SHT30 and the BME280:

The to devices on the same position and have a konstant different over 8-10°C with the SHT30.
The BME280 work right. I have it testet with two xiaomi Temp Sensors. All have the same temperature.

But have testet with 3  x SHT30 and every the sam to high temperature, the different was every 10°C...

Why ?

ascillato, 2018-10-16:
Hi,

Please, complete the troubleshooting template in order to have the information for properly help you.

--------------------------

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

--------------------------

Example of the wiki: https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#8b-adjust-the-value-of-a-sensor-and-send-it-by-mqtt

If commands `sub`, `add`, etc does not work for you means that you have an old version. Please update to last dev version from http://thehackbox.org/tasmota/

--------------------------

General Questions are better to be addressed in the Chat. Thanks

Jason2866, 2018-10-16:
Have you tried a simple arduino sketch to see if you get the same results?
I think it is not tasmota....

closed 2018-10-16 by ascillato2