arendst/Tasmota #4087: MQTT TLS Verify Fingerprint Fail

adicrescenzo, 2018-10-16:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used :
Visual Studio Code with Platformio Home 1.0.2 Core 3.6.1rc2
- [x] Hardware used : 
Sonoff Basic
- [x] Provide the output of command ``status 0`` :

2:43:00 RSL: Received Topic /status, Data Size 1, Data 0
22:43:00 RSL: Group 0, Index 1, Command STATUS, Data 0
22:43:00 RSL: stat/livingroomlights/STATUS = {"Status":{"Module":1,"FriendlyName":["Luce Soggiorno"],"Topic":"livingroomlights","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:43:00 RSL: stat/livingroomlights/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:04:11","StartupUTC":"2018-10-16T21:38:49","Sleep":0,"BootCount":1,"SaveCount":2,"SaveAddress":"FA000"}}
22:43:00 RSL: stat/livingroomlights/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-10-16T23:37:14","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
22:43:00 RSL: stat/livingroomlights/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-BiteMeBo2.4",""],"TelePeriod":300,"SetOption":["00008009","55A18000","00000000"]}}
22:43:00 RSL: stat/livingroomlights/STATUS4 = {"StatusMEM":{"ProgramSize":550,"Free":452,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAC7D4","24000000","23B617CE","00000000"]}}
22:43:00 RSL: stat/livingroomlights/STATUS5 = {"StatusNET":{"Hostname":"livingroomlights-3404","IPAddress":"192.168.1.12","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:8B:0D:4C","Webserver":0,"WifiConfig":5}}
22:43:00 RSL: stat/livingroomlights/STATUS6 = {"StatusMQT":{"MqttHost":"raspberrypi2.local","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8B0D4C","MqttUser":"livingroomlights","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:43:00 RSL: stat/livingroomlights/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 16 21:43:00 2018","Local":"Tue Oct 16 22:43:00 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:30","Sunset":"17:28"}}
22:43:00 RSL: stat/livingroomlights/STATUS10 = {"StatusSNS":{"Time":"2018-10-16T22:43:00"}}
22:43:00 RSL: stat/livingroomlights/STATUS11 = {"StatusSTS":{"Time":"2018-10-16T22:43:00","Uptime":"0T00:04:11","Vcc":3.516,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Vodafone-BiteMeBo2.4","RSSI":74,"APMac":"BC:15:AC:EB:DC:CB"}}}

I'm trying to use MQTT with TLS enabled but I'm still getting the same error:

22:48:40 WIF: Checking connection...
22:48:40 WIF: Connected
22:48:44 UPP: Multicast disabled
22:48:44 MQT: Attempting connection...
22:48:44 MQT: Verify TLS fingerprint...
22:48:45 MQT: Failed

The MQTT broker (Mosquitto) is installed on my raspberrypi2 in my local network, I've calculated the fingerprint using the command

`openssl s_client -connect raspberrypi2.local:8883 < /dev/null 2>/dev/null | openssl x509 -fingerprint -noout -in /dev/stdin`

This is my user_config.h values for this section:

#define MQTT_HOST              "raspberrypi2.local"
#define MQTT_FINGERPRINT1      "XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX"  // [MqttFingerprint1]
//#define MQTT_FINGERPRINT2      "A5 02 FF 13 99 9F 8B 39 8E F1 83 4F 11 23 65 0B 32 36 FC 07"  // [MqttFingerprint2]
#define MQTT_PORT              8883              // [MqttPort] MQTT port (10123 on CloudMQTT)
#define MQTT_USER              PROJECT           // [MqttUser] MQTT user
#define MQTT_PASS              "xxxxxxxxxxxxx"       // [MqttPassword] MQTT password

Can anyone help me please? Any help will be appreciated

ascillato2, 2018-10-16:
Hi,

TLS is a memory hungry feature. So, to make it work you need to disable a lot of features in order to free RAM. Several users had post in issues their user_config.h file as a working example. 

Please, check those. 

Thanks.

----------------------------

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-16 by ascillato2

adicrescenzo, 2018-10-17:
Can you please add some links to previous issue with a good example of user_config? I didn’t find it.

Many thanks

ascillato, 2018-10-17:
https://github.com/arendst/Sonoff-Tasmota/issues/3930

adicrescenzo, 2018-10-19:
@ascillato unfortunately this version of the user_config.h didn't solve my problem.
I'm stille getting the Verify TLS fingerprint error. Is there a way to debug the reason why the fingerprint verification is failing?

timbru31, 2018-12-02:
I'm having the same issue, even with the newest development commits. I've set the log level to `LOG_LEVEL_DEBUG_MORE`, yet the only thing I see is: `"RestartReason":"Exception"` - no exception is shown anywhere.  
Mosquitto just says: `Socket error on client <unknown>, disconnecting.`

ascillato, 2018-12-02:
Hi,

It is a "not enough memory" issue. Disable as much features as you can in order to use it.

timbru31, 2018-12-02:
Indeed, core 2.4.2 works for me!
`1543771838: New client connected from 192.168.178.41 as DVES_XX (c1, k10, u's20-1').`
Thanks a lot! ❤️

ascillato, 2018-12-02:
you are welcome!!! thanks for share your results :+1:

kleini, 2018-12-25:
Can you please give a hint, what sizes of images are small enough to run MQTT over TLS?
I am facing the same problem and I am not able to solve it using a minimal user_config.h and core 2.4.2.
I see the following sizes when compiling in PlatformIO:

    DATA:    [=====     ]  50.9% (used 41692 bytes from 81920 bytes)
    PROGRAM: [====      ]  42.0% (used 429916 bytes from 1023984 bytes)
    Uploading 434064 bytes from .pioenvs/sonoff/firmware.bin to flash at 0x00000000

player64, 2019-01-16:
The same issue here, flashed witch minimum features on Wemos d1 mini 4mb. 
Log from Mqtt server
`1547675732: Socket error on client <unknown>, disconnecting.`

majherek, 2019-01-17:
Which Core you use? 2.5.0? See #4952

player64, 2019-01-17:
Thanks for your reply. Yes, I did use 2.5.0 core. Maybe is a problem with the certificate, I generated it by using this script https://raw.githubusercontent.com/owntracks/tools/master/TLS/generate-CA.sh 
Is it a method to validate fingerprint with mosquitto server? Thanks any help will be thankful.

majherek, 2019-01-17:
> Thanks for your reply. Yes, I did use 2.5.0 core. Maybe is a problem with the certificate, I generated it by using this script https://raw.githubusercontent.com/owntracks/tools/master/TLS/generate-CA.sh
> Is it a method to validate fingerprint with mosquitto server? Thanks any help will be thankful.

1) I think there is a problem with verificating certs in Core 2.5.0.

2) What you mean by "Is it a method to validate fingerprint with mosquitto server?". This script (generate-CA.sh) is to generate CA (master cert) and after that server/client cert issued by previosly genereted CA cert (signed by this CA cert). 
If you want to check fingerprint of mosquitto cert you can use:
```
openssl s_client -connect <mqtt_server>:<port> < /dev/null 2>/dev/null | openssl x509 -fingerprint -noout -in /dev/stdin | awk -F"=" '{print $2}'|sed -e 's/:/ /g'
```