arendst/Tasmota #4088: Add the possibility to send POWER state at periodic time

schmurtzm, 2018-10-16:
It could be useful to have a parameter to force Tasmota to send on MQTT his current POWER state at periodic time.
For example a console command line like this to force the publication every 30 seconds : 
ForcePowerStatePublication 30

With Home Assistant you need to force the check of the state at start (see "[Power state on start-up without retain option](https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant)"). 
It means that to be sure to have the right state at Home Assistant start you need to make configuration on HA. :(

Thanks for your awesome work ;)

**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-10-16:
Hi,

Why you want that? There is no need to send periodically the status of relays. Tasmota sends it when they change. Also with the example (https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#power-state-on-start-up-without-retain-option) you can make HA to check all devices at once at startup.

If anyway you want to send that, you can use a rule to perform that action (using last Tasmota version).

closed 2018-10-16 by ascillato2

schmurtzm, 2018-10-16:
Just for answer to your question, you can make a simple test to reproduce the issue with home assistant : 
1 - power on the sonoff/tasmota
2 - shutdown your MQTT server
3 - switch off the sonoff
4 - restart your MQTT server

Now you're in a state where you have Sonoff powered "off" with "on" state on Home Assistant so the state is not totally reliable.
I'm going to make it works with rules but I still think that this parameter could be useful ;)

ascillato, 2018-10-16:
Hi,

I use home assistant and I have set exactly as the example (https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#power-state-on-start-up-without-retain-option)

So, If I do like you said, my Home Assistant will know the exact real status of all my Tasmotas.

schmurtzm, 2018-10-17:
I would say "no".
This example show how to refresh the state at home assistant start. In the test that I give previously I don't reboot home assistant but just the MQTT server so this automation is never executed and the status will not be exact.
A workaround is to repeat this automation at periodic time like this  :

`  alias: "Power state on HA start-up"
  trigger:
  - platform: homeassistant
    event: start
  - platform: time
    minutes: '/5'
    seconds: 00     # Check every 5 minutes
    - service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power1"
        payload: ""
# If you have more : like sonoff 4CH...
		- service: mqtt.publish
      data:
        topic: "cmnd/sonoffs/power2"
        payload: ""
`
 Not so complex but it is a workaround... ;)

benzino77, 2018-10-18:
How about that? This will report every 30 sec power information equal to teleperiod state message
```
rule1 
on system#boot do
   ruletimer1 30
endon
on rules#timer=1 do
   backlog status11; ruletimer1 30
endon
```
or other approach:
```
on system#boot do
   ruletimer1 30
endon
on rules#timer=1 do
   backlog power1;power2;ruletimer1 30
endon
```

You can use `mem1` to `mem5` to make the PowerStatePublicationPeriod changeable by user without rewriting the rule.