arendst/Tasmota #4089: TASMOTA should save the Wifi configuration after it's successfully connected to AP.

wongnam, 2018-10-17:
Hi,

I suggest that TASMOTA should allow to save the Wifi configuration after it's successfully connected to an AP. Don't allow it to save config if the connection is fail eg: wrong SSID or password.

Purpose: to avoid it to stuck in lost communication situation and we will not take control to it to bring it back.

Thanks.

Frogmore42, 2018-10-17:
Tasmota has two slots for configuration.  Different people have different requirements.  There is not one solution for them all, which is why Tasmota supports many different choices.  This is much better than the choice some products make which is you have to connect to the network to configure it for that network (which it sounds like what you are suggesting).  This makes it impossible to pre-configure the device for a different location or allow moving it from one to the other without reconfiguring it.  I have my devices configured so they work both at the house and at the barnyard.  I would have to have the same network parameters at both locations if your suggestion were implemented.

wongnam, 2018-10-17:
- I mean that when we use "Scan for wifi networks" function.
- About wifi pre-configuration as you mention above still valid, no change.

[image]

ascillato, 2018-10-17:
Sorry, those fields you are pointing are the ones that have set as default in your user_config.h file. If you do a reset 1 in the console, Tasmota will use the defaults (showed there).

closed 2018-10-17 by ascillato2

wongnam, 2018-10-17:
I just want suggest my idea to make TAMOTA will has better user experience.
Maybe I don’t explain clear enough so you could not understand me well and then close my case.