arendst/Tasmota #4090: Use rule on analog0 to dimm 37 PWM1

Wifiunterputzdose, 2018-10-17:
Hey folks,

I'm trying to dimm the pwm signal on D5 GPIO14 Sensor (37 PWM1) through a potentiometer whos connected to ADC input of the ESP-12e

Here's a picture of the device configuration in the web interface
https://imgur.com/a/GwpjXvi

I tried the example from the wiki

rule on analog#a0div10 do dimmer %value% endon

But it dont change anything on GPIO14. My idea was that is stored in the variable %value% and than I have to use the variable to dimm PWM singal on GPIO14?

If I try manually to dimm the PWM1 output with "dimmer 10" it works.

arendst, 2018-10-17:
I guess your ADC is not enabled as that's the default in the provided binaries.

Enable ADC in user_config.h and re-compile.

Wifiunterputzdose, 2018-10-17:
Hello Theo thanks for the fast Response.

What I did before is to comment out the #define USE_ADC_VCC line with // at the beginning in the user_config.h. Than made a recompile and generating a bin file.

To comment out like  //#define USE_ADC , means enabling the ADC?

Ok, how can i check btw verify that I enabled ADC? Type "Status 10" in the console of the web Interface to get the readed value of ADC?

arendst, 2018-10-17:
Yes, status 10 should reflect the ADC state:
```
10:40:51 CMD: status 10
10:40:51 MQT: stat/wemos8/STATUS10 = {"StatusSNS":{"Time":"2018-10-17T10:40:51","Switch1":"ON","Switch2":"ON","Switch3":"ON","ANALOG":{"A0":5}}}
```

closed 2018-10-17 by ascillato2

Wifiunterputzdose, 2018-10-17:
As I said before the ADC was enabled. Status 10 gives me the actual value of the adc input.

But the rule don't work >> rule on analog#a0div10 do dimmer %value% endon

I get the following response:

18:27:54 CMD: rule on analog#a0div10 do dimmer %value% endon
18:27:54 RSL: stat/WiFiUnterPutzdose/RESULT = {"Command":"Unknown"}

My target is to dimm the PWM1 output on D5 GPIO14 as a function of the adc input voltage

ascillato, 2018-10-17:
If Tasmota said command unknown using the rule command means that you are using an old version. Please update your Tasmota to the latest from http://thehackbox.org/tasmota/

ascillato2, 2018-10-17:
If you need further assistant please complete the troubleshooting template.

Wifiunterputzdose, 2018-10-17:
Thanks for your response. Its possible to read the version from the web interface?

This is the way I got the tasmota version >> I changed the branch to development on this site https://github.com/arendst/Sonoff-Tasmota than downloaded the zip file trough the "clone or download"-button and opend the folder in Atom IDE

I dont want to use pregenerated bin files from your link. I want to use my own generated bin files with ATOM because I need the possibility to define/undefine options.

ascillato, 2018-10-17:
If you need further assistant please complete the troubleshooting template

Wifiunterputzdose, 2018-10-17:
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Development/Compiler/Upload tools used :
   ATOM IDE , NodeMCU-PyFlasher-3.0-x64 to flash the generated bin-Files from ATOM IDE
- [ x] Hardware used : 
   Nodemcu LoLin V3 Board
- [ ] Provide the output of command ``status 0`` :
```
19:19:20 CMD: Status 0
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"WiFiUnterPutzdose","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:55:29","StartupUTC":"2018-10-17T17:23:51","Sleep":0,"BootCount":3,"SaveCount":13,"SaveAddress":"F7000"}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS2 = {"StatusFWR":{"Version":"6.2.1.15","BuildDateTime":"2018.10.17 18:59:47","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Chillas",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS4 = {"StatusMEM":{"ProgramSize":487,"Free":516,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000407","6050A790","248003A8","000007C4","000005C0"]}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS5 = {"StatusNET":{"Hostname":"WiFiUnterPutzdose-4073","IPAddress":"192.168.8.127","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.1","Mac":"68:C6:3A:97:AF:E9","Webserver":2,"WifiConfig":5}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_97AFE9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 17 18:19:20 2018","Local":"Wed Oct 17 19:19:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS10 = {"StatusSNS":{"Time":"2018-10-17T19:19:20","ANALOG":{"A0":552}}}
19:19:20 RSL: stat/WiFiUnterPutzdose/STATUS11 = {"StatusSTS":{"Time":"2018-10-17T19:19:20","Uptime":"0T00:55:29","POWER":"ON","Dimmer":84,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"Chillas","BSSId":"C8:14:51:05:CA:4A","Channel":4,"RSSI":54}}}
```

Wifiunterputzdose, 2018-10-17:
AH ok thanks. Sry was my first post. Ok these would make sense. But where is the line to enable the Rules?

Something goes wrong. But why? I deleted the // in the USE_RULES in user_config.h
https://imgur.com/a/MKAWmsB

Should I have else to do?

Theres also an user_config_override_sample.h. Should I place there also the #define USE_RULES

ascillato2, 2018-10-17:
If you delete or comment the line `#define USE_RULES` you are not going to have the rules feature in your firmware.

I recommend to use Tasmota as default. At least up to you solve your issues. Then from something that is working you can move to modify and experiment things.

The precompiled bins are being made from the actual codebase. The latest is http://thehackbox.org/tasmota/ Those are recommended when starting with Tasmota

Wifiunterputzdose, 2018-10-17:
Ok I tried to use the precompiled bin file -> sonoff-classsic.bin
But I have the problem to get a connection to the web interface. I see the ESP_97AFE9 network. Can't connect to 192.168.4.1 because I have a private IP Adresse, so changed the IP adresse Gateway DNS etc for the ESP_97AFE9 network but cant connect. 

Where can I find all the information to enable rules on tasmota? WHat do you mean by Tasmota as default?

ascillato, 2018-10-17:
Sonoff classic does not have rules. Use the sonoff.bin please. What features are available on each file, please read the readme.

Wifiunterputzdose, 2018-10-17:
I found the following file in the file directory : xdrv_10_rules.ino

#ifdef USE_RULES checks if RUles are defined. Rules should be work.

ascillato, 2018-10-17:
> WHat do you mean by Tasmota as default?

To use the code without being modified. Please, just download again the code and compile. Rules are enabled by default.

For use rule you need to have the following line in _user_config.h_:

`#define USE_RULES`

ascillato, 2018-10-17:
Remember that you can find us also in the chat (https://discord.gg/Ks2Kzd4)

Wifiunterputzdose, 2018-10-18:
Hi now I flashed the sonoff.bin as you recommended from here http://thehackbox.org/tasmota/ 

But in this sonoff.bin file theres no ADC enabled. Status 10 in the console dont response with the actual analog value. The release said that ADC should be enabled. Whats wrong with this bin file?

`15:26:58 CMD: Status 0
15:26:58 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:26:58 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:51","StartupUTC":"2018-10-18T14:20:07","Sleep":0,"BootCount":4,"SaveCount":13,"SaveAddress":"F7000"}}
15:26:58 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.2.1.17","BuildDateTime":"2018-10-17T16:00:45","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:26:58 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Chillas","Chillas"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
15:26:58 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","000003A0","23B617CE","00001BC0"]}}
15:26:58 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"WifiModul","IPAddress":"xx","Gateway":"xx","Subnetmask":"xxx","DNSServer":"xx","Mac":"xxx","Webserver":2,"WifiConfig":5}}
15:26:58 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_97AFE9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:26:58 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 18 14:26:58 2018","Local":"Thu Oct 18 15:26:58 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:15","Sunset":"17:55"}}
15:26:58 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-10-18T15:26:58"}}
15:26:58 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-10-18T15:26:58","Uptime":"0T00:06:51","Vcc":2.371,"POWER":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":4,"RSSI":78}}}`

ascillato, 2018-10-18:
No, There is nothing wrong.

In the releases said that ADC is disabled. USE_ADC_VCC is to use the Analog input to measure the Power supply voltage.

The only one that have ADC enabled for analog readings on the A0 pin is _sonoff_sensors.bin_

Please use that file instead.

Wifiunterputzdose, 2018-10-18:
Ok now rules and adc works.

But I got still the same problem:

If I type in the console:
rule on analog#a0div10 do dimmer %value%

I get the following response:

16:23:51 CMD: rule on analog#a0div10 do dimmer %value% endon
16:23:51 RSL: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":470,"Rules":"on analog#a0div10 do dimmer %value% endon"}

My assumption is that the Rule1 has to turn on or?

Wifiunterputzdose, 2018-10-18:
Hey I got it work :D

I didnt know that typing in "Rule 1" activate the rule

ascillato, 2018-10-18:
Great!!! Glad to know that now it is working for you :+1: