arendst/Tasmota #4091: BlitzWolf® BW-SHP6 EU

WladyX, 2018-10-17:
Hi,
Will this socket be supported on tasmota?
https://www.banggood.com/BlitzWolf-BW-SHP6-EU-Plug-Metering-Version-WIFI-Smart-Socket-220V-240V-10A-Work-with-Amazon-Alexa-p-1356981.html

Thank you very much for your support.

Frogmore42, 2018-10-17:
Until someone buys one and opens one up to confirm if it has an esp8266 and exactly how it is wired, the answer is no.

arendst, 2018-10-17:
On order. Planned shipment mid November :-(

WladyX, 2018-10-17:
Nice, keep us updated, thank you!

ghmax, 2018-10-17:
Hello,
looks like a clone of the gosund sp-111. Own a couple of those devices and flashed wiith tasmota and espurna. Working perfectly.
https://www.amazon.de/Gosund-Stromverbrauch-Fernsteurung-Schaltbare-funktioniert/dp/B07FVR7PG7

regards,
max

arendst, 2018-10-21:
Hi @ghmax when using tasmota what module type did you select? And does energy monitoring work too?

Thx in advance.

ionultd, 2018-10-21:
I have the Blitzwolf BW-SHP2 smart plug open. I see the Wifi module with can so cannot see the chip. There is as usual connection to the RW/TX. How would one confirm if it is an ESP?

ghmax, 2018-10-22:
@ionultd ... just have a look at https://github.com/arendst/Sonoff-Tasmota/wiki - your device is already supported so that should be confirmation enough.

hschroyen, 2018-10-26:
Can this link help?

https://www.bastelbunker.de/gosund-sp111-mit-tasmota/

Jason2866, 2018-10-27:
Just needs to be confirmed to be 100% sure.
At the moment 99% that BW-SHP6 EU will work because it is the same hardware as Gosund SP111
which works with settings for BlitzWolf SHP2.
So if you have the chance to buy for a good price, do it!

Jason2866, 2018-11-02:
@jobhh  Feel free to do. Everyone can do

jobhh, 2018-11-02:
@Jason2866 Thank you for giving me permission 😄 Banggood hasn't shipped yet, so it will be some time before I can.

Joking aside. I was merely stating that, assuming the BW-SHP6 is not in the RELEASENOTES.md prematurely, this ticket seems to be resolved.

mcer12, 2018-11-05:
@ghmax How did you manage to open it without breaking the casing?

ghmax, 2018-11-05:
@mcer12 there is one screw on the side which you plug into an existing socket... The postet link (https://www.bastelbunker.de/gosund-sp111-mit-tasmota/) explains everything and shows pictures.

mcer12, 2018-11-05:
@ghmax Great! Many thanks!

mcer12, 2018-11-05:
I find it questionable how small gap is between high and low voltage sides of the board. I'm no expert but is that even safe?

ascillato2, 2018-11-09:
Hi,

Closing as @ghmax has confirmed that it is working.

> that was the Blitzwolf Module. Powermonitoring is also working.

closed 2018-11-09 by ascillato2