arendst/Tasmota #4092: Automatic color change depending on the time of day.

AlexTransit, 2018-10-17:
You can make a script that will automatically publish MQTT. 
In the Morning "color/Light 153" (cold).
"Color/Light 500" (warm) in the evening.
During the day to automatically change from cold to warm.

Need to listen to the publication color/light and change the value of CT. Do not understand how to make a rule on the device.

ascillato, 2018-10-17:
How to use rules are explained in the wiki.

If you need more help on that, please find us in the Tasmota Chat at :https://discord.gg/Ks2Kzd4

AlexTransit, 2018-10-22:
You cannot do this using rules.

I made a simple script on bash
Well brother made more complicated in Python
https://gist.github.com/temoto/72cb25f924ef0e7c52a360ab600fe822

Principle is simple.
From dawn till the middle of the day (half between sunrise and sunset) is set to white. Further smoothly changes to warm before sunset.
After sunset up to half of the time of dawn is asleep warm. And then until dawn smoothly changes to white.

Can this functionality be integrated into the firmware?

ascillato, 2018-10-22:
Hi,

Sorry, will not be integrated as can be done by rules. Your idea is nice. Let me help you with the rules. Today I will be super busy but as soon as I can, I will post an idea to achive that and we can discuss it in the chat. Thanks.

AlexTransit, 2018-10-23:
> Hi, are you interested in doing this using rules?
yes.

> Which led type/device are you planning to use?
I use "Generic" or "Sonoff Led"

> The CT command works in that device? The rate of changing would be one hour or less?
CT commands work on my devices.
Now running the script. Every minute, Publicurutsit Topik on the group
"CMND/livings/CT"
CT changes according to a simple rule.

The current time I translate in minutes (CurrentMinute)
Translate in minutes sunrise and sunset (SunriseMinute and SunsetMinute )

ShiftMidleNightMinute = (((60*24)-(SunsetMinute-SunriceMinute))/2+SunsetMinute)-(60*24)
SunriseMinute = SunriseMinute + ShiftMidleNightMinute 
SunsetMinute = SunsetMinute  + ShiftMidleNightMinute 
CurrentMinute = CurrentMinute  + ShiftMidleNightMinute 
if CurrentMinute >(60*24) then
 CurrentMinute = CurrentMinute - (60*24)
end if
MidleDayMinute=((SunsetMinute -SunriseMinute)/2) + SunsetMinute

if (CurrentMinute < SunriseMinute) then
 coef = (500-153)/SunriseMinute
 CT = 500 - (CurrentMinute  * coef)
if (CurrentMinute > SunriseMinute and CurrentMinute < MidleDayMinute ) then
 CT = 153
else if (CurrentMinute > MidleDayMinute and CurrentMinute <= SunsetMinute ) then
 coef = (500-153)/(SunsetMinute-MidleDayMinute)
 CT = (SunsetMinute-MidleDayMinute) * coef + 153
else if (CurrentMinute > SunsetMinute ) then
 CT = 500
endif

ascillato, 2018-10-23:
Example:

Go to Console and type:

```
rule1 1
rule1 on system#boot do event CalcMyMidDay endon on Time#Set do event CalcMyMidDay endon on event#CalcMyMidDay do backlog var1 %sunset%; sub1 %sunrise%; mult1 0.5; endon
rule2 1
rule2 on Time#Minute do event myCT=%value% endon on event#myCT<%sunrise% do backlog scale2 %value%,%var1%,%sunrise%,500,153; CT %var2% endon on event#myCT>%sunset% do CT 500 endon on event#myCT<%var1% do CT 153 endon on event#myCT>%var1% do backlog scale2 %value%,%var1%,%sunset%,153,500; CT %var2% endon
```

Explanation:

```
rule1 1 // enable rule set 1
rule1
on system#boot do event CalcMyMidDay endon // at startup calc midday time in minutes
on Time#Set do event CalcMyMidDay endon // every hour calc midday time in minutes (useful when your Tasmota will be power on more than 24 hours)
on event#CalcMyMidDay do backlog var1 %sunset%; sub1 %sunrise%; mult1 0.5; // calculate midday var1 = (sunset - sunrise) / 2     ( example 20 hs = 1200 min, 8 hs = 480 min -> var1 = 720 min = 12 hs )

rule2 1 // enable rule set 2
rule2
on Time#Minute do event myCT=%value% endon // every minute will be executed the event myCT with actual minutes from midnight

Remember that rules in tasmota are trigger, so the first one to meet the condition will be executed leaving the rest without being executed. So the order of event#myCT triggers is very important here.

on event#myCT<%sunrise% do backlog scale2 %value%,%var1%,%sunrise%,500,153; CT %var2% endon
on event#myCT>%sunset% do CT 500 endon
on event#myCT<%var1% do CT 153 endon
on event#myCT>%var1% do backlog scale2 %value%,%var1%,%sunset%,153,500; CT %var2% endon
```

AlexTransit, 2018-10-24:
Thank you very much.
Need this in the wiki add

only this rule not change CT.
```
07:09:00 RUL: TIME#MINUTE performs "backlog event myCT=429"
07:09:00 MQT: stat/living1/RESULT = {"Event":"Done"}
07:09:00 RUL: EVENT#MYCT<%SUNRISE% performs "backlog scale2 429,198.00,489,500,153; CT 125.71"
07:09:00 RUL: EVENT#MYCT>%VAR1% performs "backlog scale2 429,198.00,1066,153,500; CT 125.71"
07:09:00 MQT: stat/living1/RESULT = {"Scale2":"115.00"}
07:09:01 MQT: stat/living1/RESULT = {"CT":500}
07:09:01 MQT: stat/living1/RESULT = {"Scale2":"125.59"}
07:09:01 MQT: stat/living1/RESULT = {"CT":500}
07:10:00 RUL: TIME#MINUTE performs "backlog event myCT=430"
07:10:00 MQT: stat/living1/RESULT = {"Event":"Done"}
07:10:00 RUL: EVENT#MYCT<%SUNRISE% performs "backlog scale2 430,198.00,489,500,153; CT 125.59"
07:10:00 RUL: EVENT#MYCT>%VAR1% performs "backlog scale2 430,198.00,1066,153,500; CT 125.59"
07:10:00 MQT: stat/living1/RESULT = {"Scale2":"113.33"}
07:10:01 MQT: stat/living1/RESULT = {"CT":500}
07:10:01 MQT: stat/living1/RESULT = {"Scale2":"125.47"}
07:10:01 MQT: stat/living1/RESULT = {"CT":500}
```

AlexTransit, 2018-10-25:
thank.
everything worked fine in the evening
in the morning did not work

I found a small error in the rules.

```
rule2 on Time#Minute do event myCT=%value% endon on event#myCT<%sunrise% do backlog scale2 %value%,%sunrise%,%var1%,500,153; event updateCT endon on event#myCT>%sunset% do CT 500 endon on event#myCT<%var1% do CT 153 endon on event#myCT>%var1% do backlog scale2 %value%,%var1%,%sunset%,153,500; event updateCT endon on event#updateCT do CT %var2% endon
```

I will test and write down the result.

AlexTransit, 2018-10-26:
Is this the correct behavior?
```
07:34:00 RUL: EVENT#MYCT<%SUNRISE% performs "backlog backlog scale2 454,494,198.00,500,153; event updateCT"
07:34:00 RUL: EVENT#MYCT>%VAR1% performs "backlog backlog scale2 454,198.00,1061,153,500; event updateCT"
07:34:00 MQT: stat/living1/RESULT = {"Scale2":"453.11"}
07:34:01 MQT: stat/living1/RESULT = {"Event":"Done"}
07:34:01 RUL: EVENT#UPDATECT performs "CT 453.11"
07:34:01 MQT: stat/living1/RESULT = {"POWER2":"ON","Dimmer":48,"Color":"1069","HSBColor":"0,0,48","Channel":[6,41],"CT":453}
07:34:01 MQT: stat/living1/RESULT = {"Scale2":"255.93"}
07:34:01 MQT: stat/living1/RESULT = {"Event":"Done"}
07:34:01 RUL: EVENT#UPDATECT performs "CT 255.93"
07:34:01 MQT: stat/living1/RESULT = {"POWER2":"ON","Dimmer":48,"Color":"5624","HSBColor":"0,0,48","Channel":[34,14],"CT":256}
```

ps
```
"Version":"6.2.1.18","BuildDateTime":"2018-10-25T16:52:49","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)

rule1 on system#boot do event CalcMyMidDay endon on Time#Set do event CalcMyMidDay endon on event#CalcMyMidDay do backlog var1 %sunset%; sub1 %sunrise%; mult1 0.5; endon

rule2 on Time#Minute do event myCT=%value% endon on event#myCT<%sunrise% do backlog scale2 %value%,%sunrise%,%var1%,500,153; event updateCT endon on event#myCT>%sunset% do CT 500 endon on event#myCT<%var1% do CT 153 endon on event#myCT>%var1% do backlog scale2 %value%,%var1%,%sunset%,153,500; event updateCT endon on event#updateCT do CT %var2% endon
```

ascillato, 2018-10-26:
I don't know, is that the behaviour you want? 

What is the odd thing you see?

AlexTransit, 2018-10-26:
```
07:34:01 MQT: stat/living1/RESULT = {"POWER2":"ON","Dimmer":48,"Color":"1069","HSBColor":"0,0,48","Channel":[6,41],"CT":453}
07:34:01 RUL: EVENT#UPDATECT performs "CT 255.93"
07:34:01 MQT: stat/living1/RESULT = {"POWER2":"ON","Dimmer":48,"Color":"5624","HSBColor":"0,0,48","Channel":[34,14],"CT":256}
```
set CT=456 and immediately CT=256

I began to understand how the rules work.
I'll fix and lay out the code.