arendst/Tasmota #4094: Serial Adapter - Termite only showing [00]s

Jallyman, 2018-10-17:
Hi All,

I am no expert at this, but I am having an issue so bear with me. I successfully flashed and used my Sonoff Pow r2 yesterday. I was able to connect to it over my Wi-Fi network.

I needed to change the device to a different Wi-Fi network and couldn't reach the device this morning as I believe the IP had changed.

This is when the issue occured:
I plugged in my serial adapter and connected over termite but am presented with a continuous steam of "[00]"s. I tried re-flashing, but it hasn't solved the problem.

Has anyone experienced this before? If I need to provide any more information let me know.

Thanks,

Jasper

arendst, 2018-10-17:
As noted in some more issues here the Pow R2 uses the serial interface to communicate with the energy monitoring chip.

The result you see is expected as it also runs on a different baudrate.

I hope you didn't connect the pow to AC while connecting the serial adapter as otherwise your serial connected PC would go up in fire...

Jallyman, 2018-10-17:
Nope, luckily I'm not crazy and have not connected mains power.

So... is there a work around? Do I need to use a different baudrate? Or is this just sitting as a known issue?

arendst, 2018-10-17:
Look through the issues here. As said the baudrate is different. Find out what it is and set termite to this baudrate. (Try 4800 baud).

You might then be able to execute serial commands. No guarantee given though.

Jallyman, 2018-10-17:
Yea, just started looking now. Much appreciated!

closed 2018-10-17 by Jallyman