arendst/Tasmota #4096: Issue with CounterDebounce

gabbermorph, 2018-10-17:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : OTA
- [x] Hardware used : Sonoff Basic / Reed
- [x] Provide the output of command ``status 0`` :
```
19:26:07 SRC: WebConsole from 192.168.177.20
19:26:07 RSL: Received Topic /status, Data Size 1, Data 0
19:26:07 RSL: Group 0, Index 1, Command STATUS, Data 0
19:26:07 MQT: stat/sonoff01/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff01"],"Topic":"sonoff01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:26:07 MQT: stat/sonoff01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T06:39:52","StartupUTC":"2018-10-17T10:46:15","Sleep":0,"BootCount":46,"SaveCount":172,"SaveAddress":"FA000"}}
19:26:07 MQT: stat/sonoff01/STATUS2 = {"StatusFWR":{"Version":"6.2.1.16","BuildDateTime":"2018-10-16T18:00:45","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
19:26:07 MQT: stat/sonoff01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["TiJay","TiJay"],"TelePeriod":300,"SetOption":["00008009","55818000","00000001"]}}
19:26:07 MQT: stat/sonoff01/STATUS4 = {"StatusMEM":{"ProgramSize":486,"Free":516,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","8FDAE394","000003A0","23B617CE","000007C0"]}}
19:26:07 MQT: stat/sonoff01/STATUS5 = {"StatusNET":{"Hostname":"sonoff01","IPAddress":"192.168.177.41","Gateway":"192.168.177.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.177.1","Mac":"5C:CF:7F:98:5D:CB","Webserver":2,"WifiConfig":5}}
19:26:07 MQT: stat/sonoff01/STATUS6 = {"StatusMQT":{"MqttHost":"hassio","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_985DCB","MqttUser":"homeassistant","MqttType":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:26:07 MQT: stat/sonoff01/STATUS7 = {"StatusTIM":{"UTC":"Wed Oct 17 17:26:07 2018","Local":"Wed Oct 17 19:26:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"07:52","Sunset":"18:22"}}
19:26:07 MQT: stat/sonoff01/STATUS10 = {"StatusSNS":{"Time":"2018-10-17T19:26:07","COUNTER":{"C1":60464}}}
19:26:07 MQT: stat/sonoff01/STATUS11 = {"StatusSTS":{"Time":"2018-10-17T19:26:07","Uptime":"0T06:39:52","Vcc":3.154,"POWER":"OFF","Wifi":{"AP":1,"SSId":"TiJay","BSSId":"C0:25:06:C0:62:45","Channel":1,"RSSI":58}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

I have still the same issue on Slow rotations on my Gasmeter with the reed contact.
On Slow Rotations it counts twice, even with debounce on 3200.

And what means Counter1n?

Its actually running on a USB adapter 550mA through a FTDI on 3.3v.

Cheers Flo

ascillato, 2018-10-17:
If the signal has noise it will be counted more times than the real value.

CounterN means that the gpio will be configured without internal pull up resistor

ascillato2, 2018-10-17:
I use the counter for a water metering and have no issues there. The output of your gas counter uses 3.3V or 5V? Seems to be something in your circuit. Have you tried to add a pull up or pull down (depending on how your sensor is connected)?

gabbermorph, 2018-10-17:
Its a simple reed contact in glass, and its klicking when a magnet passes. Just like a windowcontact.
I am using a sonoff basic powered over a FTDI adapter on USB. And its set to 3.3V

The sensor is connected to GPIO14.

ascillato, 2018-10-17:
Ah, ok, so you need a pull up resistor and your reed need to be connected to GND

gabbermorph, 2018-10-17:
oh wow, okay that makes sense, does it work with 3.3v on the resistor?

ascillato, 2018-10-17:
Yes,

you can use something like:

```
3.3V -----|
          |
          R = 10K
          |
GPIO -----|----------------- REED ----- GND

```

ascillato, 2018-10-17:
For reeds or switches, if you are not using a pull up, you will have a floating state that can induce noise.

gabbermorph, 2018-10-17:
man that helped and i leaned from it, will throw that for testing on my breadboard.

Thanks!

ascillato, 2018-10-17:
You are welcome.

Please, close this issue and if you experiece any problem, reopen it again. Thanks

ascillato2, 2018-10-17:
You are welcome.

Please, ask to reopen this issue if you experience any problem related to this. Thanks. 

Remember that you can find us also in the Tasmota Chat

closed 2018-10-17 by ascillato2

gabbermorph, 2018-10-22:
So, 
i build up a Pull Up Resistor, but there is no change. It still counts twice on slow ratations.
CounterDebounce is on 3200

gabbermorph, 2018-10-22:
So, 
i added a pull up resistor and still the same behavior, it counts twice on slow rotations.
Is it possible to let it count only on CLOSE and not on OPEN after the Debouncecounter is gone?

ascillato, 2018-10-23:
https://github.com/arendst/Sonoff-Tasmota/blob/a7c912111cd78048170a9b4934dd0dd808f86663/sonoff/xsns_01_counter.ino#L81

Counter is measuring the falling edge, so that means that if you have double measurements, means that you have noise or bad pull up or pull down resistors.