arendst/Tasmota #4098: Sonoff S20 in startup loop and press button no effect

hamwong, 2018-10-18:
I upgrade my Sonoff S20 then upgrade to 6.2.1.15 from 5.x and I accidently typo in console,
I got 2 wireless, 1=Ham and 2=HamIot, usually SSID1=HamIot, SSID2=Ham (SSID2 for backup purpose), Wificonfig=5, as 2 SSID in differnce subnet, today I typo SSID1 to SSID 1, it make SSID1 fail to access and I also not seeing it into SSID2 network, then I unplug the S20 for few second, then power on again, then I still unable to access, I found heavy packet loss by ping, I use Serial and found plug is in boot loop cycle,
It seens it success to SSID2, get IP from router, then restart again, problem solved by "ERASE Flash - ALL Flash Contents" in ARDUINO, another issue is long press power button won't reset config, as I never try before, I was wonder, why set it to long press 40 sec for factory restore?

```
00:00:00 Project sonoff Sonoff-s20-1 (Topic Sonoff-s20-1, Fallback DVES_9F3E20, GroupTopic sonoffs) Version 6.2.1.15-2_4_2
00:00:00 WIF: Connecting to AP2 Ham in mode 11N as Sonoff-s20-1-7712...
00:00:05 WIF: Connected
17:13:38.827 -> 00:00:05 DNS: Initialized
17:13:38.827 -> 00:00:05 HTP: Web server active on Sonoff-s20-1-7712.local with IP address 192.168.1.59

17:13:39.569 -> Exception (28):
17:13:39.569 -> epc1=0x4022cb20 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000001 depc=0x00000000
17:13:39.569 -> 
17:13:39.569 -> ctx: cont 
17:13:39.569 -> sp: 3ffff8b0 end: 3fffffd0 offset: 01a0
17:13:39.569 -> 
17:13:39.569 -> >>>stack>>>
17:13:39.569 -> 3ffffa50:  feefeffe 3ffffb43 3ffffa80 feefeffe  
17:13:39.569 -> 3ffffa60:  feefeffe feefeffe feefeffe 4022b1d0  
17:13:39.569 -> 3ffffa70:  3ffffba0 3ffe8304 3ffffb00 4022bb5d  
17:13:39.569 -> 3ffffa80:  00000005 3ffffcd8 3ffffab0 feefeffe  
17:13:39.569 -> 3ffffa90:  3ffe8304 feefeffe feefeffe 3ffffb00  
17:13:39.569 -> 3ffffaa0:  4022b1d0 3ffffba0 3ffffcd8 4022bf18  
17:13:39.569 -> 3ffffab0:  00000005 feefeffe feefeffe 3fff5a24  
17:13:39.569 -> 3ffffac0:  3ffe8304 3ffe8304 3ffffb70 4022b478  
17:13:39.569 -> 3ffffad0:  00000000 3fff505c 00000000 00000000  
17:13:39.569 -> 3ffffae0:  00000005 00000011 fe302073 3fff5060  
17:13:39.569 -> 3ffffaf0:  00000000 3ffe8304 3ffffba0 4022b634  
17:13:39.569 -> 3ffffb00:  00000000 ffffffff 00000000 00000000  
17:13:39.569 -> 3ffffb10:  00000005 00000005 fe302073 feefeffe  
17:13:39.569 -> 3ffffb20:  feefeffe 3fff505c 00000000 feefeffe  
17:13:39.569 -> 3ffffb30:  3ffffc70 3ffffc60 0000000c 00000005  
17:13:39.569 -> 3ffffb40:  00ff5a35 00000011 3ffffbf0 4022b478  
17:13:39.569 -> 3ffffb50:  feefeffe feefeffe 00000000 00000014  
17:13:39.569 -> 3ffffb60:  3ffffca0 3ffffc90 00000014 00000005  
17:13:39.569 -> 3ffffb70:  3fff505f 00000005 00000002 ffff0208  
17:13:39.569 -> 3ffffb80:  3ffffc90 00000013 3ffffc10 00000200  
17:13:39.569 -> 3ffffb90:  4025e388 3ffe8304 00000200 4022ae49  
17:13:39.569 -> 3ffffba0:  3fff0f29 feefeffe 000001fa ffff0208  
17:13:39.569 -> 3ffffbb0:  3fff0f24 000001ff 0000000c 3fff368c  
17:13:39.569 -> 3ffffbc0:  00000020 00000020 00000000 40100575  
17:13:39.569 -> 3ffffbd0:  3ffe8304 3ffe8304 3ffffc60 00000021  
17:13:39.569 -> 3ffffbe0:  3fff368c 00000392 00000392 4010020c  
17:13:39.569 -> 3ffffbf0:  4025e3b0 00000011 3fff5a24 401006dc  
17:13:39.569 -> 3ffffc00:  00000010 00000010 00000012 40228b18  
17:13:39.569 -> 3ffffc10:  3fff368c 3fff0f24 3fff505c 4022ae8c  
17:13:39.569 -> 3ffffc20:  3ffffca0 3ffffc90 0000000c 401004e4  
17:13:39.569 -> 3ffffc30:  00000010 3ffffca8 0000000a 00000200  
17:13:39.569 -> 3ffffc40:  4025e388 3fff0f24 3fff505c 40229081  
17:13:39.569 -> 3ffffc50:  3ffffca0 3ffffc90 0000000c 7fffffff  
17:13:39.569 -> 3ffffc60:  3ffffd10 00000000 3ffffccc 3ffffd78  
17:13:39.569 -> 3ffffc70:  00000000 3fff1980 3fff0f24 402290b8  
17:13:39.569 -> 3ffffc80:  3ffffca0 3ffffc90 0000000c 40208a7e  
17:13:39.569 -> 3ffffc90:  7474714d 6c627550 44687369 3ffffcd8  
17:13:39.569 -> 3ffffca0:  00000001 3fff1124 00000000 00000000  
17:13:39.569 -> 3ffffcb0:  00000000 3fff1980 3ffffd10 4021153f  
17:13:39.569 -> 3ffffcc0:  00000000 00000000 00000000 00000000  
17:13:39.569 -> 3ffffcd0:  00000000 00000000 3a4c5352 00000020  
17:13:39.569 -> 3ffffce0:  3ffffd74 00000154 3f302073 00000092  
17:13:39.569 -> 3ffffcf0:  00000000 3fff19fc 3ffffd10 4021172b  
17:13:39.569 -> 3ffffd00:  00000000 00000002 00000005 402117c4  
17:13:39.569 -> 3ffffd10:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd20:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd30:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd40:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd50:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd60:  ffffffff ffffffff ffffffff ffffffff  
17:13:39.569 -> 3ffffd70:  00ffffff 4f464e49 00000031 3fff1124  
17:13:39.569 -> 3ffffd80:  3ffe8640 3fff180c 3fff1980 402290b8  
17:13:39.569 -> 3ffffd90:  3ffffdb0 3ffffda0 0000000c 40100f22  
17:13:39.569 -> 3ffffda0:  00000005 00000000 00000020 3fff1124  
17:13:39.569 -> 3ffffdb0:  3ffe8640 3fff180c 3fff1980 402117eb  
17:13:39.569 -> 3ffffdc0:  40101c3d 3ffeda98 3ffee508 40214e9e  
17:13:39.569 -> 3ffffdd0:  3fff1c31 005e785a 3ffee508 40101e0e  
17:13:39.569 -> 3ffffde0:  3ffea8d4 00000000 00000000 00000100  
17:13:39.569 -> 3ffffdf0:  fffffff4 005e785a 4010224e 00000100  
17:13:39.569 -> 3ffffe00:  7fffffff 3ffea8d4 3ffea8d4 00000001  
17:13:39.569 -> 3ffffe10:  00000001 00000040 00002200 4000050c  
17:13:39.569 -> 3ffffe20:  3fffc278 005e785a 00002200 4000050c  
17:13:39.569 -> 3ffffe30:  3fffc278 40101fe8 3fffc200 00000022  
17:13:39.569 -> 3ffffe40:  3ffea8e0 00000030 00000010 ffffffff  
17:13:39.569 -> 3ffffe50:  402176dc 00000030 0000001c ffffffff  
17:13:39.569 -> 3ffffe60:  402176b4 00000001 0000000a 3fff1820  
17:13:39.569 -> 3ffffe70:  00000000 00000000 0000001f 40104bfd  
17:13:39.569 -> 3ffffe80:  00000000 00000000 0000001f 00000003  
17:13:39.569 -> 3ffffe90:  3fff1980 3fff180c 3fff1980 4021a510  
17:13:39.569 -> 3ffffea0:  4020b6e5 00000030 00000010 ffffffff  
17:13:39.569 -> 3ffffeb0:  4020b6f1 00000000 00000000 00000062  
17:13:39.569 -> 3ffffec0:  00000000 00000000 000003da fffffffe  
17:13:39.569 -> 3ffffed0:  00000000 3fffc6fc 00000000 00000014  
17:13:39.569 -> 3ffffee0:  00000034 00000004 3fff35e0 00000030  
17:13:39.569 -> 3ffffef0:  3ffe8cf0 3fff3448 00000030 401004e4  
17:13:39.569 -> 3fffff00:  40225338 3fff3448 3fff5080 4021fa0e  
17:13:39.669 -> 3fffff10:  00000000 4bc6a7f0 18d4fdf3 00000000  
17:13:39.669 -> 3fffff20:  402525da 3fff184c 3fff180c 00000001  
17:13:39.669 -> 3fffff30:  4025379b 00000000 4bc6a7f0 00000003  
17:13:39.669 -> 3fffff40:  3fff1980 3fff180c 3fff173c 4021a648  
17:13:39.669 -> 3fffff50:  00000000 00000000 3fff184c 4021aa94  
17:13:39.669 -> 3fffff60:  001bf8d4 0000001c 00000000 3fff35e0  
17:13:39.669 -> 3fffff70:  3fff18a0 00000000 00000000 3fff35e0  
17:13:39.669 -> 3fffff80:  3ffe865c 3fff18a4 0000192b 402054b7  
17:13:39.669 -> 3fffff90:  3fffdad0 0000001c 000000fa 3fff35e0  
17:13:39.669 -> 3fffffa0:  3fffdad0 00000000 3fff18a4 4021ab6e  
17:13:39.669 -> 3fffffb0:  3fffdad0 00000000 3fff35d9 40228d5c  
17:13:39.669 -> 3fffffc0:  feefeffe feefeffe 3ffe86ac 40100739  
17:13:39.669 -> <<<stack<<<

17:13:39.703 ->  ets Jan  8 2013,rst cause:1, boot mode:(3,7)
17:13:39.703 -> 
17:13:39.703 -> load 0x4010f000, len 1384, room 16 
17:13:39.703 -> tail 8
17:13:39.703 -> chksum 0x2d
17:13:39.703 -> csum 0x2d
17:13:39.703 -> vbb28d4a3
17:13:39.703 -> ~ld
```

Frogmore42, 2018-10-18:
It seems @hamwong has solved the issue with a full erase already.

The reason the 40 second button press didn't work is the device was rebooting before 40s was up.  If the question is why use 40s, that is because you don't want someone to accidentally activate the reset.

When a device has some bad configuration, it can be hard to recover from it.

ascillato2, 2018-10-18:
Hi,

Closing issue as the question has been answered. Thanks

closed 2018-10-18 by ascillato2

hamwong, 2018-10-18:
> It seems @hamwong has solved the issue with a full erase already.
>
> The reason the 40 second button press didn't work is the device was rebooting before 40s was up. If the question is why use 40s, [...]

I think 10-15 sec is long enough to avoid accident, also I was wonder why need to do full flash for upgrade? shouldn't be invisgate why S20 in loop cycle after upgrade? It's all original hardware without mod (I mean add extra sensor)

Frogmore42, 2018-10-18:
There is a lot of software involved in making Tasmota work.  Most of that software is completely outside of Tasmota's control.  The error could be anywhere, but it is likely to be in the area that very few people understand and that is completely undocumented.  The Wi-Fi code comes from Espressif SDK and when you upgrade from one version to the next or back it is likely there could be issues.  I don't believe you will find anyone here who has the capability or the time to figure it out, especially when a complete erase of flash makes the problem go away.

OTA under the best of circumstances is very difficult to get right every time in every case.  When you add OTA on top of software that was not designed with that use case in mind, you will get results like this.  Tasmota is leveraging the functionality provided by the Arduino project, which is leveraging the functionality provided by the espressif SDK, which is a mix of open source and closed source code.  I doubt there is a single person who understands exactly how it all works together.

I have one device that is in a somewhat hard to access place.  It is still running the same version I loaded before I installed it.  It is now many months old.  I am pretty certain that if I tried to OTA update it, it would no longer work.  My choice is simple, leave it working as is, or take it down and spend half a day getting it back to working with new software.  The choice is simple 😀

In your case, you entered bad configuration information and it didn't work.  It is not a great user experience, but Tasmota is not a commercial product and even many of them would do bad things if given bad data.

digiblur, 2018-10-18:
@Frogmore42 very well put! Might very well be Wiki worthy somewhere and a save for a copy/paste later.

hamwong, 2018-10-18:
thanks @Frogmore42 from what you said, now I understand issue my happens because wifi module been upgarde and cause issue, never mind, normal kit is easy to fix, I got one more Hufan fail after upgarde, that is headache to recover, but power delta is a good function for tracking the power usage, it will take few more days to do it then