arendst/Tasmota #4100: Shelly 1 is not working with official precompiled binaries

daniode, 2018-10-18:
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Development/Compiler/Upload tools used : plattformio 1.0.2. 3.6.1rc3 for selfcompiled, esptool 0.4.12
- [x] Hardware used : Shelly 1 v1 (delivered yesterday), FTDI Adapter
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Moinsen!

I have some trouble with a shelly 1. Compliling and flashing with platformio is working fine. Flashing the provides precompiled binaries from http://thehackbox.org/tasmota/ recommendes at issue https://github.com/arendst/Sonoff-Tasmota/issues/4050 with esptool is working fine too. But both are not working properly. The output on tty is like [0C][00][0C][04][0C][0C][04][0C][04][00]$\`rc[1F][02][02]\[10]"[1B][04][0C][02][0C][04][0C][04][0C][0C][04][0C]#[0C] ... error codes?

If i flash the edited 6.1.1 image from the hook up (his changes are IMHO part of tasmota now) the device will answer properly on tty, is accpeting my commands and joining my wifi.  

Any hints? 
regards
danio

ascillato, 2018-10-18:
Hi, shelly is fully supported only in last version of the development branch.

Please download from http://thehackbox.org/tasmota/

ascillato2, 2018-10-18:
You have a baudrate misconfiguration. please, use command baudrate and set the same as your tty

closed 2018-10-18 by ascillato2

daniode, 2018-10-18:
Thanks for your Answer. I had already tried the linked firmware as i wrote my request. i tried it several times with and without erasing (incl. a 2mb image full of zeros).
And it is not about the baudrate. I tried some different rates. no effect. it was not accepting my command.
btw it does not open an AP.
Maybe my equiptment is broken but it works fine with the modified 6.1.1 and stock firmware. 
Now, i have three shelly left...:) lg danio