arendst/Tasmota #4106: Add more switches

lovinggrey, 2018-10-19:
I want to use 8 switches to send mqtt messages to homeassistant through the 8266 module. I wonder if it can be implemented in Tasmota?

ascillato, 2018-10-19:
Hi,

It is already implemented. You can select 8 switches right now in module configuration

ascillato, 2018-10-19:
[image]

ascillato, 2018-10-19:
May be you have a very old version of Tasmota.

Please upgrade to last.

You can use the precompiled bins from http://thehackbox.org/tasmota/ to be up to the last changes!

-----------------------------------

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

lovinggrey, 2018-10-19:
> Hi,
> 
> It is already implemented. You can select 8 switches right now in module configuration

I found it, but there are only 11 GPIO available, and 8 switches and 8 entities are not enough.

ascillato, 2018-10-19:
What do you meant by entities?

You said before:

> I want to use 8 switches to send mqtt messages

So, that is possible and already implemented. Sorry, I don't understand what is the issue?

lovinggrey, 2018-10-19:
I want to use switches to send on and off messages to homeassistant to automate other devices, such as turning on and off lights.

ascillato, 2018-10-19:
Great!

That is implemented and you can do that

ascillato, 2018-10-19:
May be you are not finding the information in the wiki.

here is an example on how to do that:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#16-using-an-external-button-with-single-press---double-press-and-hold

lovinggrey, 2018-10-19:
I tried to add eight switches, but mqtt could only send power on or power off, without messages like Power1 on or power2 on after adding multiple entities.

lovinggrey, 2018-10-19:
http://v.youku.com/v_show/id_XMjk0NTk3MzIxMg==.html?spm=a2hzp.8244740.0.0

The device in the video was made by me, but I couldn't write firmware. The firmware was written by a former friend who wrote botton instead of switch. So it's not too good to use. It may take two times to control the switch.

ascillato, 2018-10-19:
> I tried to add eight switches, but mqtt could only send power on or power off, without messages like Power1 on or power2 on after adding multiple entities.

Please, read the example I gave you. It uses command publish and you can send any type of mqtt message you want.

If you have physically a pushbutton instead of a button, please, see in the examples also is explained that.

It is all configuration, no need to modify Tasmota.

lovinggrey, 2018-10-19:
I'll check the information again. If there are any problems, I'll ask you again. Thank you very much.

ascillato, 2018-10-19:
No problem.

Remember that you can find us in the Tasmota Chat at: https://discord.gg/Ks2Kzd4

General questions are faster and better to be addressed there. Thanks.

ngalfas, 2019-01-01:
I recently stumbled upon the 8 'input' switches use case.  I only wanted to have 8 inputs from physical switches. So, i added 8 switch items on the configuration page on Tasmota.  I didn't need to drive any relay (was using Wemos D1R2).

- Without any relay configured, the SwitchTopic returns  plain POWER status without POWER1, POWER2 etc.    

- In order to use the SwitchTopic with many input switches for MQTT purposes YOU HAVE TO ADD A RELAY on the configuration!    I know this is written somewhere in the wiki but it doesn't make any sense!   

I saw somewhere else on the issues, someone thought that by having 8 "input" switches he had to configure 8 relays onboard so that 16 GPIO's were needed in that case scenario.  Can you blame his order of thoughts?

It's a little mystery to me why it's necessary to have at least one relay in order to have MULTIPLE POWERx MQTT messages.