arendst/Tasmota #4108: Error in Sensor-Status-JSON-String through wrong comma

Fab333, 2018-10-19:
**Describe the bug**
Hey Guys,
I think i found an error regarding a received JSON-String.
My actual setup is, that I configured my tasmota-device with an WS2812B-LED-Stripe, MPR121 and a BMP280. 
If I now send an Mqtt-request to ”cmnd/sonoffDevice/STATUS 10”, I do receive the requested sensor-data on Mqtt-Topic “stat/sonoffDevice/STATUS10” in a JSON-String.

The JSON-String looks like this:
`{"StatusSNS":{"Time":"2018-10-19T13:09:52","BMP280":{"Temperature":24.0,"Pressure":1014.2},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0,},"TempUnit":"C"}}`

The Problem is: When I try to parse these JSON-String **there is an error because of the “,” after “,"Button12":0”**.

I fixed these Error through modifying the file “xsns_30_mpr121.ino” in line 364 through line 369. 
I replaced the following code:
```
if (FUNC_JSON_APPEND == function) 
{
	snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s\"Button%i\":%i,"), mqtt_data, j, BITC(i, j));
}
```

With this one:
```
if (FUNC_JSON_APPEND == function) 
{
	if (j > 0)
	{
		snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s,\"Button%i\":%i"), mqtt_data, j, BITC(i, j));
	} else
	{
		snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("%s\"Button%i\":%i"), mqtt_data, j, BITC(i, j));
	}
}
```

That fixed the Error in the JSON-String.
First of all: Can someone confirm this error?
Is there a better Way to fix it?

Best Regards
Fab

P.S. Sorry if I violated any Issue-Report-Protocols. I was an Issue-Reporting-Virgin until just now. :)

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [x] _Searched the problem in issues and in the wiki_
- [x] _Hardware used_ : NodeMcu DevKit  
- [x] _Development/Compiler/Upload tools used_ : ArduinoIDE 1.8.5
- [-] _Provide the output of command_``status 0`` : I would submit if necessary. I already flashed my device with the mentioned workaround so i hope it is not neccessary.
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_
Wrote everything in "**Describe the bug**"

**Expected behavior**
_A clear and concise description of what you expected to happen._
Expected a JSON-String like this:
`{"StatusSNS":{"Time":"2018-10-19T13:09:52","BMP280":{"Temperature":24.0,"Pressure":1014.2},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0},"TempUnit":"C"}}`
(Without a comma after the Last Button-State)

**Screenshots**
_If applicable, add screenshots to help explain your problem._
I think the Problem is clear without a Screenshot. :)

ascillato, 2018-10-19:
Hi,

Status 0 is necessary. I don't know which Tasmota version did you use and which config. Please share it.

Fab333, 2018-10-19:
Hey,

Sorry, here is Status 0:
`17:00:18 MQT: stat/FlexLampe/STATUS = {"Status":{"Module":18,"FriendlyName":["FlexLampe"],"Topic":"FlexLampe","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:00:18 MQT: stat/FlexLampe/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T03:38:48","StartupUTC":"2018-10-19T12:21:30","Sleep":0,"BootCount":3,"SaveCount":7,"SaveAddress":"3F5000"}}
17:00:18 MQT: stat/FlexLampe/STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018.10.19 14:07:00","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
17:00:18 MQT: stat/FlexLampe/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MySSID",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
17:00:18 MQT: stat/FlexLampe/STATUS4 = {"StatusMEM":{"ProgramSize":491,"Free":532,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":0,"Features":["00000407","0F80A7B4","00000000","00B4049E","00000002"]}}
17:00:18 MQT: stat/FlexLampe/STATUS5 = {"StatusNET":{"Hostname":"FlexLampe-1195","IPAddress":"192.168.0.32","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:13:44:AB","Webserver":2,"WifiConfig":2}}
17:00:18 MQT: stat/FlexLampe/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1344AB","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:00:18 MQT: stat/FlexLampe/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 19 16:00:18 2018","Local":"Fri Oct 19 17:00:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:16","Sunset":"17:53"}}
17:00:18 MQT: stat/FlexLampe/STATUS10 = {"StatusSNS":{"Time":"2018-10-19T17:00:18","BMP280":{"Temperature":23.7,"Pressure":1013.8},"MPR121A":{"Button0":0,"Button1":0,"Button2":0,"Button3":0,"Button4":0,"Button5":0,"Button6":0,"Button7":0,"Button8":0,"Button9":0,"Button10":0,"Button11":0,"Button12":0,},"TempUnit":"C"}}
17:00:18 MQT: stat/FlexLampe/STATUS11 = {"StatusSTS":{"Time":"2018-10-19T17:00:18","Uptime":"0T03:38:48","Vcc":3.026,"POWER":"OFF","Dimmer":10,"Color":"191919","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"MySSID","RSSI":70,"APMac":"34:31:C4:E8:67:78"}}}`

I also added the status to my first entry.

ascillato, 2018-10-19:
Thanks.

I ask you the Tasmota Version because we had fixed recently a bug like that, but this is another one. Thanks for reporting :+1:

arendst, 2018-10-19:
Just updated and released.