arendst/Tasmota #4112: Sonoff Basic Garage Door is unable to open or close manually when MQTT Broker is down

rtv73, 2018-10-19:
Hello,

I am using a Sonoff Basic 6.2.1 with my garage door and integrated with OH. It works great, both with the app via OH and via the old way using the wall garage switch on the wall.

My issue is that when is that when my MQTT server is unreachable, the switch on the wall is unable to open the door. It will start, but within a second or so it stops. It seems like the relay gets trigger by the firmware within 1 second.

20:43:19 CMD: status 0
20:43:19 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["GarageDoor1"],"Topic":"GarageDoor1","ButtonTopic":"GarageDoor1","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:43:19 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:35:20","StartupUTC":"2018-10-19T19:07:59","Sleep":0,"BootCount":68,"SaveCount":252,"SaveAddress":"F8000"}}
20:43:19 RSL: STATUS2 = {"StatusFWR":{"Version":"6.2.1","BuildDateTime":"2018-09-09T16:50:26","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:43:19 RSL: STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":2,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["MaXi WiFi",""],"TelePeriod":300,"SetOption":["00008081","55818000","00000000"]}}
20:43:19 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","000003A0","23B617CE","00000000"]}}
20:43:19 RSL: STATUS5 = {"StatusNET":{"Hostname":"GarageDoor1-5033","IPAddress":"192.168.1.31","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:A7:93:A9","Webserver":2,"WifiConfig":5}}
20:43:19 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 19 19:43:19 2018","Local":"Fri Oct 19 20:43:19 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"07:16","Sunset":"17:53"}}
20:43:19 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-10-19T20:43:19","Switch2":"OFF"}}
20:43:19 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-10-19T20:43:19","Uptime":"0T00:35:20","Vcc":3.182,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MaXi WiFi","RSSI":84,"APMac":"1C:87:2C:35:3B:A8"}}}

ascillato, 2018-10-19:
Hi,

Are you using switchtopic for the reed sensors?

If yes, that is the problem. Switchtopic are for switches and they have a fallback to control the relays when there is no mqtt.

For the reed sensor, it is better to use a rule:

go to console and type:

```
rule1 1
rule1 on switch1#state do publish cmnd/GarageDoor1/switch1 %value% endon
```

This rule will do similar to switchtopic but without the fallback feature. Please adapt it to your needs. More examples are in the wiki at rules.

ascillato, 2018-10-19:
Yes, you need to type switchtopic 0 in order to let the rule work.

ascillato2, 2018-10-19:
### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-19 by ascillato2

rtv73, 2018-10-20:
No luck yet...  SwitchTopic is 0, and rule1 is enabled but sama behavior:

```
02:56:46 CMD: SwitchTopic

02:56:46 RSL: stat/GarageDoor1/RESULT = {"SwitchTopic":""}

02:57:39 CMD: rule1

02:57:39 RSL: stat/GarageDoor1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":445,"Rules":"on switch1#state do publish cmnd/GarageDoor1/switch1 %value% endon"}
```

ascillato, 2018-10-20:
Please, type weblog 4 in the console to have more debug information and make your reed to change states. Please, share the output.

rtv73, 2018-10-20:
Here you are.... The door was open, and it close... when I tried to open it again it gets stocked within a 1sec as the issue I am trying to resolved when the MQTT is offline:

`20:29:31 CMD: SwitchTopic 0

20:29:31 MQT: stat/GarageDoor1/RESULT = {"SwitchTopic":""}

20:30:11 CMD: WebLog 4

20:30:11 MQT: stat/GarageDoor1/RESULT = {"WebLog":4}

20:30:11 CFG: Saved to flash at FA, Count 298, Bytes 3584

20:30:21 WIF: Checking connection...

20:30:21 WIF: Connected

20:30:41 WIF: Checking connection...

20:30:41 WIF: Connected

20:31:01 WIF: Checking connection...

20:31:01 WIF: Connected

20:31:02 SRC: Switch

20:31:02 MQT: stat/GarageDoor1/RESULT = {"POWER":"OFF"}

20:31:02 MQT: stat/GarageDoor1/POWER = OFF

20:31:08 SRC: Switch

20:31:08 MQT: stat/GarageDoor1/RESULT = {"POWER":"ON"}

20:31:08 MQT: stat/GarageDoor1/POWER = ON

20:31:09 CFG: Saved to flash at F9, Count 299, Bytes 3584

20:31:21 WIF: Checking connection...

20:31:21 WIF: Connected`

ascillato, 2018-10-20:
Ok, that means that your rule is not being executed.

In your previous test, do you had switchtopic to 0 ?

Switchtopic overrides rules

rtv73, 2018-10-20:
Yes, SwitchTopic 0 is in the first line of the log.

ascillato, 2018-10-21:
Ah ok. Sorry.

That is weird. Should work by now. In my devices that config works fine. Please, can you try to update to last Tasmota from http://thehackbox.org/tasmota/ ?