arendst/Tasmota #4113: Find3 integration possible

damylen, 2018-10-19:
Has anyone looked into integration of Find3 scanner inside Tasmota? Having many tasmota's around the house this could really benefit a good internal positioning solution

Esp client:
https://github.com/DatanoiseTV/esp-find3-client

Find3 info:
https://www.internalpositioning.com/doc/

ascillato, 2018-10-19:
Hi,

It is a nice idea, but will make your esp8266 to be in both modes AP and STATION consuming a lot more of energy due to the sleep function can not be used. Also, a lot of CPU power will be used managing both connections and several lags in the normal smart switch behaviour should occur.

I think, IMHO, that should be better to have a specific device and firmware for that.

andrethomas, 2018-10-19:
I agree with @ascillato - This is not viable given that Tasmota is already running very tight code. Of course, you're welcome to fork the codebase and implement this on your own.

damylen, 2018-10-19:
alright, thank you for your comments

closed 2018-10-19 by damylen