arendst/Tasmota #4119: DisplayText Commands

HillNet, 2018-10-21:
All,
I appreciate this work is in development and there is limited to no support at this point in tme.

However, if anyone has time to guide me I would appreciate it. Thank you!

I have my OLED up and displaying on DisplayModel 2 and DisplayMode 2, but I'd like to play around the the DisplayText commands as written up here: [https://github.com/arendst/Sonoff-Tasmota/wiki/DisplayText-Command](url)

My two issues are:
1. I seem to can't find what the "p" does, eg `[f1p7x0y5]` or `[s1p21c1l01]`, and
2. When sucessfully using a rule to display my sensors, the %value% seems to get written over without clearing the original value making it impossible to read the current value. 

Any insights would be appreciated and I really like the display support in this wonderful piece of software.

--
Regards,

Roland

ascillato, 2018-10-21:
Please, find us in the Tasmota Chat. Thanks

Jason2866, 2018-10-21:
Hello,
to use this commands you have to set DisplayMode 0
https://github.com/arendst/Sonoff-Tasmota/wiki/DisplayText-Command
Caution! Not all commands are implemented still work in progress

HillNet, 2018-10-21:
Thanks @Jason2866.

I have Display Mode 0 set and can get 4 rows of sensor data displaying. 

I'll have to look in using [z] in the rule as new values aren't clearing old ones.

closed 2018-10-21 by ascillato2