arendst/Tasmota #4122: AC Triac 0-Z Dimmer for Tasmota?

HoerMirAuf, 2018-10-21:
Hi there,

is there a possibilty to get a Function i Tasmota for thie device?

[https://arduinodiy.wordpress.com/2018/01/23/dimming-an-ac-lamp-via-mqtt/](https://arduinodiy.wordpress.com/2018/01/23/dimming-an-ac-lamp-via-mqtt/)

So GPIO14 and the LED GPIO Port by example could be used for that ...

Would be nice!

andrethomas, 2018-10-21:
Should work if you set GPIO pin to PWM mode.

Have a look in the Wiki here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands and search for PWM -You will find all the pwm related settings and commands there.

ascillato, 2018-10-21:
Sorry, will not work. Tasmota can't manage the zero crossing. You will have firing at different times of the AC mains signal and you can burn your lamp.

That is why you need a separate mcu to manage the firing. Tasmota is not a real time software and can't manage the firing. 

This was discussed already in issues.

closed 2018-10-21 by ascillato2