arendst/Tasmota #4123: Shelly 1 Switchmodes

HorstRado, 2018-10-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : platformio
- [x] Hardware used : Shelly 1
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```
Hello,

I need help for Shelly1 with Tasmota. I have installed Tasmota 6.2.1.x
I want to replace an eltaco with the Shelly. Therefor I connect the pushbutton to the
SW pin of the Shelly. I thought switchmode 0 would work like a pushbutton on a sonoff:
Pressing once: relay is on, press again: relay is off. But in my case the relay goes on when 
I press the pushbutton and go off when the button is released. Same with switchmode 1. All other
switchmodes do not work at all. Any idea?

Regards
Horst

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-10-21:
Switchmode 0,1 and 2 work like a normal switch: relay off when open, relay on when closed (or inverted)

To use a bushbutton you need to select switchmode 3 or 4.

See #3989 too

ascillato2, 2018-10-21:
Hi

You said version 6.2.1.x but which one. Several issues are solved using last version. Please, update to last version and test again please.

HorstRado, 2018-10-21:
Thanks. The Problem is solved with latest development version 6.2.1.18

closed 2018-10-21 by HorstRado