arendst/Tasmota #4125: Sonoff Pow loses Energy Total values

FabianHM, 2018-10-21:
Hello,
Why does the Sonoff Pow lose its Total value already after a day without mains voltage? I'm sure that was not the case in the older version.

Thanks for your Ideas,
Greetings Fabian

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Development/Compiler/Upload tools used : Sonoff-Tasmota 6.2.0 
- [x] Hardware used : Sonoff Pow (V1)

ascillato, 2018-10-21:
Please, update to the last version from http://thehackbox.org/tasmota/

Frogmore42, 2018-10-21:
The pow and pow r2 share a lot of code, so it is not surprising they would have the same issue, that is likely fixed.

ascillato2, 2018-10-21:
When you have time, please told us if it works for you.

closed 2018-10-21 by ascillato2

anwar3000, 2020-06-03:
> Hello,
> Why does the Sonoff Pow lose its Total value already after a day without mains voltage? I'm sure that was not the case in the older version.
>
> Thanks for your Ideas,
> Greetings Fabian
>
> * [x] [...]

SAME PROBLEM WITH ME 
8.3.1(sensors)