arendst/Tasmota #4127: serial comm not working properly after flashing , sonoff basic

gnkarn, 2018-10-21:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues) , yes
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting) yes
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers) yes
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)  ,, invite is not working
- [ x] Development/Compiler/Upload tools used : platformio
- [ x] Hardware used : sonoff basic
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
00:00:00 APP: Boot Count 17
00:00:00 APP: Set Baudrate to 43200

no mater what is the baud rate set , after flashing it shows the message of 43200 , and after that , just garbage .
tried with all different baud rates ,  tried , with different arduino versions ( supplied with tasmota) , 

the green led on the board , stays on all the time after , sonoff restart .

any ., tips  of what to try are appreciated , i have many of this sonoff basic , and is the first time i was not able to flash one.

using the latest version , on development.

tis
**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-10-21:
Seems that you are having issues compiling or flashing. Try to use the precompiled bins from http://thehackbox.org/tasmota/

closed 2018-10-21 by ascillato2

arendst, 2018-10-21:
For some reason the settings area is corrupt. Pls erase the flash and try again.

gnkarn, 2018-10-21:
thanks @arendst !, tried reseting flash using esptool , 
then flashing again:

> ecking size .pioenvs/sonoff/firmware.elf
> Building .pioenvs/sonoff/firmware.bin
> Memory Usage -> http://bit.ly/pio-memory-usage
> DATA:    [======    ]  57.6% (used 47152 bytes from 81920 bytes)
> 
> PROGRAM: [====      ]  35.5% (used 363472 bytes from 1023984 bytes)
> Configuring upload protocol...
> Looking for upload port...
> Use manually specified: /dev/tty.usbserial-AM01RMWQ
> Uploading .pioenvs/sonoff/firmware.bin
> Uploading 367616 bytes from .pioenvs/sonoff/firmware.bin to flash at 0x00000000
> ................................................................................ [ 22% ]
> ................................................................................ [ 44% ]
> ................................................................................ [ 66% ]
> ................................................................................ [ 89% ]
> .......................................                                          [ 100% ]
> ========================= [SUCCESS] Took 54.31 seconds =========================
>  
> ================================== [SUMMARY] ==================================
> Environment sonoff              [SUCCESS]

but now the message on the console is 
00:00:00 APP: Boot Count 0
00:00:00 APP: Set Baudrate to 306000 
then garbage 

and after flashing the led remains on , all the time

it is really very strange ,

gnkarn, 2018-10-21:
what worked  was (1) to reset flash , (2)then load precomp bin, reset config  ,(3) after that compile my code and load the bin  from the web interface .

Jason2866, 2018-10-22:
esptool  erase_flash didnt work seetings survived...