arendst/Tasmota #4130: POW module send MQTT message out of Telemetry period

AlexBoosh, 2018-10-21:
POW module send MQTT message out of Telemetry period.
Telemetry periot set 30 second
Periodically messages arrive outside this period.
In the log below, these are messages in 19:19:53 and 19:19:54.
Message in 19:19:53 have wrong Power value.

Can help me?
Thanks.
Log:
```
19:19:12 MQT: HA/pow/STATE = {"Time":"2018-10-19T19:19:12","Uptime":"0T13:56:02","Vcc":3.464,"POWER":"ON","Wifi":{"AP":1,"SSId":"iptv3","RSSI":64,"APMac":"F8:32:E4:B1:D0:08"}}
19:19:12 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:19:12","ENERGY":{"Total":7.961,"Yesterday":0.000,"Today":7.961,"Period":12,"Power":1457,"Factor":1.00,"Voltage":217,"Current":6.168}}
19:19:42 MQT: HA/pow/STATE = {"Time":"2018-10-19T19:19:42","Uptime":"0T13:56:32","Vcc":3.463,"POWER":"ON","Wifi":{"AP":1,"SSId":"iptv3","RSSI":64,"APMac":"F8:32:E4:B1:D0:08"}}
19:19:42 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:19:42","ENERGY":{"Total":7.973,"Yesterday":0.000,"Today":7.973,"Period":12,"Power":1459,"Factor":1.00,"Voltage":217,"Current":6.158}}

19:19:53 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:19:53","ENERGY":{"Total":7.978,"Yesterday":0.000,"Today":7.978,"Power":0,"Factor":0.00,"Voltage":217,"Current":6.151}}
19:19:54 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:19:54","ENERGY":{"Total":7.978,"Yesterday":0.000,"Today":7.978,"Power":1455,"Factor":1.00,"Voltage":217,"Current":6.158}}

19:20:12 MQT: HA/pow/STATE = {"Time":"2018-10-19T19:20:12","Uptime":"0T13:57:02","Vcc":3.463,"POWER":"ON","Wifi":{"AP":1,"SSId":"iptv3","RSSI":64,"APMac":"F8:32:E4:B1:D0:08"}}
19:20:12 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:20:12","ENERGY":{"Total":7.985,"Yesterday":0.000,"Today":7.985,"Period":12,"Power":1451,"Factor":1.00,"Voltage":217,"Current":6.151}}
19:20:42 MQT: HA/pow/STATE = {"Time":"2018-10-19T19:20:42","Uptime":"0T13:57:32","Vcc":3.463,"POWER":"ON","Wifi":{"AP":1,"SSId":"iptv3","RSSI":64,"APMac":"F8:32:E4:B1:D0:08"}}
19:20:42 MQT: HA/pow/SENSOR = {"Time":"2018-10-19T19:20:42","ENERGY":{"Total":7.998,"Yesterday":0.000,"Today":7.998,"Period":12,"Power":1455,"Factor":1.00,"Voltage":217,"Current":6.158}}
```

ascillato, 2018-10-21:
Hi,

Sorry that you didn't follow the troubleshooting template. You would find that this issue has been addressed before. You have set a powerdelta set.

Please, review the commands in the wiki and also search in the issues. 

Thanks. 

Remember that you can find us also in the Tasmota Chat.

closed 2018-10-21 by ascillato2

AlexBoosh, 2018-10-22:
Thank you, Excuse for troubling.
All work fine, thx!