arendst/Tasmota #4131: Help for rules

riddik14, 2018-10-21:
hi I need help, I can not set the rules. then my intention is to turn on a bathroom vacuum cleaner with the condition that if the sensor dht 22 measures humidity greater than 70 lights for 10 minutes and then turns off.
Apart from the rule the SonOff must still be independent of the rule, so I tried these commands but it does not work as I say, it turns on and does not turn off

RULE1 on AM2301#Humidity<70.0 do POWER ON endon

Ruletimer2 2

RULE2 on POWER ON do Rules#Timer=2 POWER OFF endon

Rule1 1
RULE2 1

ascillato, 2018-10-21:
Hi,

Sorry your syntax is incorrect. You need to follow as explained in the wiki. 

The trigger for the power state is:
On Power1#state=1 do.... 

Please review the wiki. 

Also remember that you can find us also in the Tasmota Chat. Thanks

closed 2018-10-21 by ascillato2

riddik14, 2018-10-21:
I have try your correction. Don’t work. If humidity is >70% power go on, but don’t go to off why?
 I use SonOff base with dht22 on gpio14

curzon01, 2018-10-22:
First delete yours
```
rule1 "
rule2 "
```

Then
```
rule1 on AM2301#Humidity>70 do POWER ON endon on AM2301#Humidity<68 do POWER OFF endon
rule1 1
rule1 4
```
This contains a [hysteresis](https://en.wikipedia.org/wiki/Hysteresis) of 3%
Hysteresis recommended on sensors otherwise your device can flap when the sensor value vary around the switchpoint.

riddik14, 2018-10-22:
The code 

rule1 on AM2301#Humidity>70 do POWER ON endon on AM2301#Humidity<68 do POWER OFF endon
it works with the values ​​of the dht sensor, but if I turn on the bathroom's suction sonoff, it will turn on and off immediately. so the question is: in addition to the rule with the condition, it is possible to turn on manually with 10-15 minutes timing excluding the sensor rule? Thanks u

curzon01, 2018-10-23:
That's described under [Wiki Rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) Examples

- read and understand [9. Simple Thermostat Example](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#9-simple-thermostat-example)
- adapt sensor to your needs (replace termo with your humdity)
- adapt setpoint upper/lower limits to your humidity limits