arendst/Tasmota #4133: WS2812: using color2 will always power on the LEDs

makefu, 2018-10-21:
Hey, i am not sure if this is a bug or feature request.

**Is your feature request related to a problem? Please describe.**
I am currently struggling with the ws2812 functionality of the tasmota. I have a status LED which gets turned on by a motion detection sensor using WS2812 on GPIO2 and Switch1 on GPIO4.

I would like to change the Color which is being used when Motion is detected, however when i use Color2 command the LEDs get powered on even though they were powered off before:

This is the trace of mosquitto_sub:
```
/ham/flurlicht/cmnd/POWER OFF # i turn the leds off
stat/flurlicht/RESULT {"POWER":"OFF"}
stat/flurlicht/POWER OFF
/ham/flurlicht/cmnd/Color2 36eaff # i set the new color
stat/flurlicht/RESULT {"POWER":"ON","Dimmer":57,"Color":"1E8591","HSBColor":"186,79,57","Channel":[12,52,57]} # result is the leds being turned on
```

I have set the topic to `/ham/%topic%/%prefix%/` however this does not seem to be used by the LED commands but this is another issue i guess.

**Describe the solution you'd like**
For my use case to work either the default behavior of color2 should be changed to automatically starting the LEDs or to have an option where to toggle this behavior.

**Describe alternatives you've considered**
I thought about subscribing to the POWER topic and quickly toggle `OFF`after i've set the led colors, however this results in the leds flickering and turning off again.

**Additional context**

Keep up your awesome work!

ascillato, 2018-10-22:
Hi,

The command color2 powers up the led in the color you have selected but keeping last dimming value. The last part is the only difference with color command. 

What you have experimented is the expected behaviour of the command, as explained in the wiki at commands. 

So, if you want that when your motion detector triggers, your led goes on with a specific color different than the one being used at that moment, you can set a rule.

Example: (adapt to your requirements)

Go to the console and type:

Rule1 1 
Rule1 on switch1#state=1 do color2 1,0,0 endon

makefu, 2018-10-22:
@ascillato i guessed that, however i want to be able to change the next trigger color. My use case is the following:
1. a script regularly (every 15 minutes) checks google for events in my calendar and retrieves the respective calendar color
2. -> If there is an event in the near future, set the color of the LEDs to the color of the event
2. -> If there is no event, set the LEDs to default color

I used the script before with my own software which was not turing on the LEDs when receiving a new color but simply store the value for the next trigger of the PIR sensor. I was looking into getting the same behavior via tasmota.

Source code of the script: http://cgit.euer.krebsco.de/ampel/tree/ampel/google_muell.py

It is nice to be able to set another color on trigger, however i want to be able to do exactly this externally.

ascillato, 2018-10-23:
Ok, can be done with rules.

For your switch:

```
Rule1 1
Rule1 on switch1#state=1 do color2 %mem1% endon
```

And for your automation software. Just send the following MQTT message to set the next color:

`cmnd/your_sonoff_topic/MEM1 #YOUR_COLOR`

closed 2018-10-23 by ascillato2

makefu, 2018-10-23:
neat, thanks!