arendst/Tasmota #4134: Automatic color change depending on the time of day.

AlexTransit, 2018-10-22:
You cannot do this using rules.

I made a simple script on bash
Well brother made more complicated in Python
https://gist.github.com/temoto/72cb25f924ef0e7c52a360ab600fe822

Principle is simple.

between sunrise and (sunset - sunrise )/2 - set white (CT=153)
between (sunset - sunrise )/2 and sunset - every 2 minute change CT from 153 to 500
between sunset and (sunrise - sunset)/2 - set warm (CT=500)
between (sunrise - sunset)/2 and sunrise - every 2 minute change CT from 500 to 153

Can this functionality be integrated into the firmware?

Jason2866, 2018-10-22:
This will not implemented even if not possible via rules. 
ESP devices have very limited ressources...
A great Job for a higher level System  -> Home Automation 😉

Jason2866, 2018-10-22:
May i overlook something i see nothing why it cant be done with rules.

ascillato, 2018-10-22:
Hi, duplicated issue.

closed 2018-10-22 by ascillato2