arendst/Tasmota #4137: Chirp! moisture sensor

madmicio, 2018-10-22:
I looked for the iusse but I did not find the answer I'm looking for. is it possible to use this humidity sensor with tasmota?
https://www.tindie.com/products/miceuz/i2c-soil-moisture-sensor/

Which capacitive humidity sensors are supported?

Which rain sensors are supported?

andrethomas, 2018-10-22:
None from what I know.

Soil moisture sensors are tricky because the cheap chinese ones generally last a few days and then you start not getting useful data due to oxidation or corrosion on the probes.

I think adding I2C support for that device is the least of the problem but perhaps someone like @mike2nl will entertain it.

Just my 2c ;)

mike2nl, 2018-10-22:
@madmicio 
there are more then one sensor to messaure humidity and temperature.
There is a DHT11, DHT22, AM2302 and some others.
Depending on the quality you need and what you will messaure.

Right now there is no RAIN sensor supported. Mostly used in combination with weather station.
But we have now the TX20 wind direction and speed.

So possible you can give us some more information what you precisely will do.

madmicio, 2018-10-22:
my project is to insert the humidity sensor into the ground and activate irrigation only when the humidity falls below a certain value. I bought the sensor in the link because it is difficult to corrode.

with the rain sensor instead I would like to change the time of my alarm in case of rain

ascillato, 2018-10-23:
Hi,

There is already an open request for this sensor at #2065

Let's continue the thread there.

Thanks

closed 2018-10-23 by ascillato2

madmicio, 2018-10-23:
Why did you close? Beyond the CHIRP, I asked if there are rain sensors and other soil moisture sensors supported.

andrethomas, 2018-10-23:
Your question was answered, so no need for it to remain open.