arendst/Tasmota #4138: Access Web environment after a few days

lscosta90br, 2018-10-22:
After a few days I can not access my web environment by ip

```
14:07:35 MQT: stat / garage-control-wireless / STATUS = {"Status": {"Module": 18, "FriendlyName": "Garage Control" PowerShell ": 0," PowerShell ": 0," PowerShell ": 0," PowerShell ": 1," SaveState ": 1, : 0}
```

Is there any command to take the web environment out of power save?

**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

**Describe the bug**
_A clear and concise description of what the bug is._

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_
- [ ] _Searched the problem in issues and in the wiki_
- [ ] _Hardware used_ : 
- [ ] _Development/Compiler/Upload tools used_ :
- [ ] _Provide the output of command_``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2018-10-22:
Hi, there is not in Tasmota such thing as web page power save.

Seems that your device just hungs or you are having wifi issues with your router setup.

andrethomas, 2018-10-22:
Also try using `wificonfig 5` so that your device does not restart when losing wifi connectivity

andrethomas2, 2018-10-22:
Closing this issue as it has been answered. 

If you have further questions please do not hesitate to follow one of the provided support entry points listed below.

### Support Information

See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.

closed 2018-10-22 by andrethomas2

lscosta90br, 2018-10-23:
How do I schedule a reboot every day at 11 o'clock?

ascillato, 2018-10-23:
go to timers. set your time in timer1 and put rule1 as output

go to console:

```
rule1 1
rule1 on Clock#Timer=1 do restart 1 endon
```

For more information, please visit the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki)