arendst/Tasmota #4141: wemos D1 mini every 5 min restart

rt400, 2018-10-23:
**IMPORTANT NOTICE**
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Development/Compiler/Upload tools used : Arduino
- [X] Hardware used : wemos D1 mini + 4 relay board + PZEM004T
- [X] Provide the output of command ``status 0`` :
```
12:08:52 MQT: stat/3PhazesMonitors/STATUS = {"Status":{"Module":18,"FriendlyName":["3PhazesMonitors","Sonoff2","Sonoff3","Sonoff4"],"Topic":"3PhazesMonitors","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.254:9999/data/firmwares/sonoff.bin","RestartReason":"Hardware Watchdog","Uptime":"0T00:00:29","StartupUTC":"2018-10-23T09:08:23","Sleep":0,"BootCount":1695,"SaveCount":1807,"SaveAddress":"F5000"}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS2 = {"StatusFWR":{"Version":"6.2.1.18","BuildDateTime":"2018-10-21T18:00:59","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Mejahez",""],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS4 = {"StatusMEM":{"ProgramSize":486,"Free":516,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","000083A0","23B617CE","00003BC0"]}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS5 = {"StatusNET":{"Hostname":"3PhazesMonitors","IPAddress":"192.168.0.91","Gateway":"192.168.0.138","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.254","Mac":"84:F3:EB:B3:63:33","Webserver":2,"WifiConfig":5}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.254","MqttPort":1883,"MqttClientMask":"3PhazesMonitors","MqttClient":"3PhazesMonitors","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 23 09:08:52 2018","Local":"Tue Oct 23 12:08:52 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3,"Sunrise":"09:23","Sunset":"19:46"}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS10 = {"StatusSNS":{"Time":"2018-10-23T12:08:52","ENERGY":{"Total":70.793,"Yesterday":2.603,"Today":0.890,"Power":89,"ApparentPower":130,"ReactivePower":95,"Factor":0.68,"Voltage":232,"Current":0.560},"DHT11":{"Temperature":28.9,"Humidity":50.0},"TempUnit":"C"}}
12:08:52 MQT: stat/3PhazesMonitors/STATUS11 = {"StatusSTS":{"Time":"2018-10-23T12:08:52","Uptime":"0T00:00:29","Vcc":2.726,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"Mejahez","BSSId":"80:2A:A8:54:8A:49","Channel":11,"RSSI":76}}}
```

hi ,
i see in info that every couple mins he made a software restrart.. i cant figure why...???

**(Please, remember to close the issue when the problem has been addressed)**

rt400, 2018-10-23:
@andrethomas how can i check the timing out ?

ascillato, 2018-10-23:
Seems that you are having some incorrect data left on the flash.

Please, try to full erase the flash using esptool.py (explained in the wiki) and then flashing again.

rt400, 2018-10-23:
@ascillato  i erase all when update to 2.4.2

ascillato, 2018-10-23:
Have you tried the precompiled bins from http://thehackbox.org/tasmota/ ?

ascillato2, 2018-10-23:
@rt400 have you tried the connections as @arendst has told you?

ascillato2, 2018-10-24:
Hi,

Have you managed to solve your issue?

rt400, 2018-10-24:
@ascillato ok i thinks that problem slove for now .
what i did is flash again the firmware include erase all , move the ping to RX TX on WEMOS .
now its seems ok and not make restart again

closed 2018-10-24 by rt400

andrethomas, 2018-10-24:
Awesome @rt400 Thanks for reporting back.