arendst/Tasmota #4142: Sonoff Dual Module GPIO1+GPIO3

zibous, 2018-10-23:
I am using a Sonoff Dual and would like to connect a DTH22 sensor. If I use Sonoff Dual R2 modules, then I can see the sensor's data but not switch the relay.

Is it possible to activate the "Serial IN / OUT" on the Sonoff Dual?

[image]

[image]

[image]

arendst, 2018-10-23:
The Sonoff Dual uses the serial interface to switch the relays so you cannot use TX/RX for connecting a sensor.

Sonoff Dual R2 uses direct connected relays and the RX/TX can be used for sensors.

Your device as shown on the picture is a Sonoff Dual and NOT a Sonoff Dual R2.

ascillato, 2018-10-23:
Hi,

In a sonoff dual, the relays are being controlled by another MCU and the esp8266 control that MCU using the serial interfase. So, it is not possible to use the serial interfase in a sonoff dual.

closed 2018-10-23 by ascillato2

zibous, 2018-10-23:
@theo:
Thanks, i understand, therefore I should use the GIPO14 for my Sonoff Dual, only I find only directly at the esp8266 control

arendst, 2018-10-23:
Yes. correct. By request of some users these GPIO's were made configurable as they seemed to be very handy in connecting wires to the chip. ;-)