arendst/Tasmota #4143: Sonoff Basic: GPIO14 to follow relay (giving 3.3 when power on, 0v when power off)

cisky78, 2018-10-23:
Hello Guys,

everithing is in the title! BTW I'd like to connect the GPIO14 to an external 3.3 V LED, which will have to act exactly as the on board LED, does anyone knows if it's possible and how? I've tried several configuration, the best one found so far is to set GPIO14 as Relay2 and then use home assistant to turn it on when the replay (1) is on, and viceversa, which I don't know if it's feasable and, anyway, whould be a workaround!

thanks for your support ;)

ascillato, 2018-10-23:
Go to console and type:

Rule1 1 
Rule1 on Power1#state do power2 %value% endon

ascillato2, 2018-10-23:
Remember that you can find us also in the Tasmota Chat.

Also, all the information is on the wiki. 

Thanks

closed 2018-10-23 by ascillato2

cisky78, 2018-10-23:
thanks a lot, i've tried, but, the issue i see is that even is i see relay 1 and 2 toggle correctly by Toggle1, the voltage from GPIO14 is 0V when ON and 3.3V when OFF, i'd like to have GPIO14 giving 3.3V when Relay1 is ON...

ascillato, 2018-10-23:
ok,

go to module config in the Tasmota webmenu and select relay2i instead of the relay2 you have selected.

the i is inverted.

cisky78, 2018-10-23:
Thanks again, this fixed all issues! i spent 2 days trying to figure out how to achive this result!